realsenseai/librealsense #11974: Export Color stream from L515 prerecorded *.bag file

chhayank-sri, 2023-07-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |L515 | 
| Firmware Version                   | (Open RealSense Viewer --> 2.53.1 | 
| Operating System & Version |   Win (10)| 
| Platform                                 | PC|
| Language                            |  MATLAB| 
| Segment			|  Robot                          | 

### Issue Description
I have recorded data using a L515 camera and generated a *.bag file. 

### Can the color stream be extracted (to a mp4 or any other format ) from the bag file?
 I have uploaded the bag file in realsense viewer which show the relevant information
[image]
**

chhayank-sri, 2023-07-10:
Hi, thank you for your answer. 
I had used rs-convert before but rs-convert gives black and white images as color channel data (screenshot attached below).
[image]

the command used was rs-convert -i test.bag -p png-path
Is there a command for getting color images back as output from rs-convert? Or any other method.

Also, as shown in the viewer screengrab the color stream does have the color information.  

I am new to using this product, so apologize for any reparation !!

MartyG-RealSense, 2023-07-10:
The only other RealSense bag-frame image extraction tool that I know of is one created by a RealSense user called **rs_bag2image**.   This tool outputs depth frames uncolored.  I do not know though whether the RGB frames are also monochrome.  

https://github.com/UnaNancyOwen/rs_bag2image

chhayank-sri, 2023-07-11:
Thank you @MartyG-RealSense, I was able to solve this!! 
rs_bag2image didn't actually help, but I wrote my own logic for reading YUYV image in L515.

Thank you for you help!!

closed 2023-07-11 by chhayank-sri

MartyG-RealSense, 2023-07-11:
You are very welcome.  It's great to hear that you created a solution, @chhayank-sri - thanks very much for the update!