realsenseai/librealsense #11975: Retrieve Depth frames from a recorded video.

Susheelabhat, 2023-07-07:
----------------------------------------------------------------------------------------------------

| Required Info                       

| Camera Model                           D435i/D455
| Firmware Version                       
| Operating System & Version    Ubuntu 17)   
| Kernel Version (Linux Only)                                            
| Platform                                     Raspberry Pi
| SDK Version                               2.0                          
| Language                                   python
| Segment			            others      

### Issue Description

I have recorded Depth and RGB frames from RealSense and dumped into .avi files. 

```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

color_path = 'V00P00A00C00_rgb.avi'
depth_path = 'V00P00A00C00_depth.avi'
colorwriter = cv2.VideoWriter(color_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)
depthwriter = cv2.VideoWriter(depth_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640, 480), 1)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

        colorwriter.write(color_image)
        depthwriter.write(depth_colormap)

        cv2.imshow('Stream', depth_colormap)

        if cv2.waitKey(1) == ord("q"):
            break
finally:
    colorwriter.release()
    depthwriter.release()
    pipeline.stop()
```

**Now i have to extract depth_frames from dumped depth video **in python**. I need the depth information from the recorded video file.** How can i retrieve it? 

my use case: For the depth frame, I need the corresponding RGB frame as well from the dumped .avi video.

MartyG-RealSense, 2023-07-07:
Hi @Susheelabhat  It will be difficult to retrieve depth information once it has been recorded to an .avi video and a lot of the depth information is likely to have been lost in the conversion.  It would be better to record to a **.bag** file, which is like a video recording of camera data and so behaves like a live camera when used by a script as the data source instead of a real camera.  A bag can record both the color and depth streams into it so that they match up when played back.

If you have not used a bag file before then you can test it by downloading a sample bag from the link below and then drag and dropping it into the center panel of the RealSense Viewer tool to automatically activate playback.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

*******

If your project requires that .avi video files are used then the RealSense SDK has a 'virtual camera' interface called **software-device** for feeding data such as video into the SDK.  It is best suited to use with the C++ language though and has had past problems when used with Python.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

*******

Another approach would be to record a bag file and then use the SDK's **rs-convert** tool to extract depth and color frames from it as PNG image files or as text-based **.csv** that can be imported into a spreadsheet or database.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

Susheelabhat, 2023-07-07:
Thanks for the info @MartyG-RealSense . The problem with bag files were that they are too huge. For 10 mins of recording, the file size was around 1GB. So is there a way to reduce this?

Also, can I use frames from .bag file as cv Mat and input it to a ML model? or in general perform openCV Operations on those .bag frames?

majinsoft, 2023-07-07:
@Susheelabhat have you tried the app [Camera 3D](https://apps.microsoft.com/store/detail/camera-3d/9NCCZNRX8QGC)?
Can record in realtime to .h3d format that is a compressed stream containing RGB, Depth and metadata. There is an utility that can extract frames and metadata from the .h3d format to standard formats (.jpg, 16 bit .png, .xml, etc..). Worth trying it if can meet your requirements. There is a free Player app for .h3d files too.

MartyG-RealSense, 2023-07-07:
The only way to reduce bag file size would be to use a lower resolution and /or FPS speed.

If you set a bag file in a script as the data source instead of a live camera then you should be able to use it with a RealSense OpenCV script, like the examples at the link below.

https://dev.intelrealsense.com/docs/opencv-wrapper

@majinsoft  Thanks very much for your advice to @Susheelabhat  :)

Susheelabhat, 2023-07-07:
@majinsoft Thanks for the suggestion, will try Camera 3D as I haven't tried it before.
@MartyG-RealSense , I see that this script is in c++, is there Python examples as well?

MartyG-RealSense, 2023-07-07:
@Susheelabhat  Python example programs that make use of OpenCV are here:

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

This includes an example called **read_bag_example.py**

Susheelabhat, 2023-07-10:
Thanks @MartyG-RealSense. That was useful. 
Also, can I know how to change fps of the recording in realsense-viewer?

MartyG-RealSense, 2023-07-10:
The FPS that is used in the bag recording will be the one that was set before streaming was enabled.  Once a stream is started then you cannot change the FPS until it is stopped.

In regards to playing back a bag that has already recorded, you cannot change the FPS in the bag but you may be able to alter the playback rate by only using every 'nth' frame (e.g every 5th frame out of 30 when using 30 FPS to simulate 6 FPS).  So in the case of a target of 3 FPS, every 10th frame could be looked at. A Python script that demonstrates this technique can be found at https://github.com/IntelRealSense/librealsense/issues/3169

MartyG-RealSense, 2023-07-16:
Hi @Susheelabhat  Do you require further assistance with this case, please?  Thanks!

Susheelabhat, 2023-07-17:
No @MartyG-RealSense . My questions are answered. Thanks for the assistance!

MartyG-RealSense, 2023-07-17:
You can set 6 FPS on the D435i by expanding open the options of the **Stereo Module** section of the options side-panel of the Viewer using the small arrow beside 'Stereo Module', then selecting an FPS from the **Frame Rate (FPS)** drop-down menu.

6 FPS is not supported on D455 though.  You can only select 5 FPS as the minimum.

[image]

closed 2023-07-24 by MartyG-RealSense