realsenseai/librealsense #11977: librealsense+realsense_ros Jetson Orin nano

MichaelKerley, 2023-07-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |   Nvidia Jetson Orin nano   |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Jetpack 5 (Ubuntu 20.04)  Jetson Linux 35.3.1 | 
| Kernel Version (Linux Only)    |  (5.1)  | 
| Platform                                 | NVIDIA Jetson Orin Nano |
| SDK Version                            |  { 2.54.1 } | 

### Issue Description

I am trying to get librealsense working with realsense_ros so that I can display a pointcloud in rviz...
I am following the instructions provided here... 
https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy
**realsense ros was installed using the method outlined in step 2 of the above instructions**

To install realsense sdk I used two different methods from this guide
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

**1: When using the apt-get method**

michael@ubuntu:~/catkin_ws$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/michael/.ros/log/c18c36e2-1cfa-11ee-a3e2-c76eb876b7b1/roslaunch-ubuntu-77199.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.144.166.95:39847/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
[... 244 lines cut ...]
[ INFO] [1688766546.022268070]: Device FW version: 05.15.00.02
[ INFO] [1688766546.022455180]: Device Product ID: 0x0B07
[ INFO] [1688766546.022794805]: Enable PointCloud: Off
[ INFO] [1688766546.022983738]: Align Depth: Off
[ INFO] [1688766546.023158591]: Sync Mode: Off
[ INFO] [1688766546.023425031]: Device Sensors: 
[ INFO] [1688766546.098159107]: Stereo Module was found.
[ INFO] [1688766546.143228487]: RGB Camera was found.
[ INFO] [1688766546.143381931]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1688766546.143445677]: num_filters: 0
[ INFO] [1688766546.143479246]: Setting Dynamic reconfig parameters.
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1688766552.828000424]: Done Setting Dynamic reconfig parameters.
/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/michael/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-1] process has died [pid 67778, exit code 127, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/michael/.ros/log/c18c36e2-1cfa-11ee-a3e2-c76eb876b7b1/camera-realsense2_camera_manager-1.log].
log file: /home/michael/.ros/log/c18c36e2-1cfa-11ee-a3e2-c76eb876b7b1/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/michael/.ros/log/c18c36e2-1cfa-11ee-a3e2-c76eb876b7b1/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2023-07-08:
Hi @MichaelKerley  Versions 2.50.0 of the librealsense SDK and 2.3.2 of the ROS wrapper are the correct pairing.

It appears that you are experiencing the same error message when installing librealsense with either apt-get or an RSUSB source code build.

`/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/michael/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev`

Ideally, there should only be one type of librealsense installation on the same computer (either package or source code, but not both).  

If the ROS wrapper was installed from source code with Step 2, then let's first try removing the librealsense package build so that only the source code build of the SDK remains.  Inputting the command below into the Ubuntu terminal should uninstall all RealSense related packages.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

If uninstalling the package version of librealsense does not make a difference, a RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/issues/2467#issuecomment-1268417031 shares the method for how they resolved their own **undefined symbol: _ZN2cv3MatC1Ev** error.  They were using a Jetson AGX Orin in their particular case.

ammarazab1, 2023-07-08:
This issues is occured because you are download ddynamic_reconfigure multiple time
make this command 
`roscd ddynamic_reconfigure`

and if it in the any workspace delete it and rebuild the workspace

and then check and remove it from main ROS distribute

`sudo apt remove ros-distro-ddynamic_reconfigure`

and reinstall ddynamic_reconfigure

MichaelKerley, 2023-07-10:
I managed to get realsense viewer working after doing
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

However now when building realsense ros I get

michael@ubuntu:~/catkin_ws$ catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Base path: /home/michael/catkin_ws
Source space: /home/michael/catkin_ws/src
Build space: /home/michael/catkin_ws/build
Devel space: /home/michael/catkin_ws/devel
Install space: /home/michael/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/michael/catkin_ws/build"
####
####
#### Running command: "make -j6 -l6" in "/home/michael/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target _realsense2_camera_generate_messages_check_deps_Metadata
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target _realsense2_camera_generate_messages_check_deps_DeviceInfo
[  0%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[  0%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target _catkin_empty_exported_target
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target diagnostic_msgs_generate_messages_nodejs
[  0%] Built target diagnostic_msgs_generate_messages_eus
[  0%] Built target diagnostic_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target diagnostic_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target tf_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target std_srvs_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[... 64 lines cut ...]
compilation terminated.
compilation terminated.
In file included from /usr/include/boost/range/iterator_range_core.hpp:38,
                 from /usr/include/boost/lexical_cast.hpp:30,
                 from /opt/ros/noetic/include/ros/transport_hints.h:34,
                 from /opt/ros/noetic/include/ros/subscribe_options.h:33,
                 from /opt/ros/noetic/include/ros/node_handle.h:42,
                 from /opt/ros/noetic/include/ros/ros.h:45,
                 from /opt/ros/noetic/include/image_transport/publisher.h:38,
                 from /opt/ros/noetic/include/image_transport/image_transport.h:38,
                 from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:8,
                 from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/usr/include/boost/range/functions.hpp:18:10: fatal error: boost/range/begin.hpp: No such file or directory
   18 | #include <boost/range/begin.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:76: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:89: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o] Error 1
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:63: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1547: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Invoking "make -j6 -l6" failed

MichaelKerley, 2023-07-10:
I managed to get realsense viewer working after doing
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

However now when building realsense ros I get

michael@ubuntu:~/catkin_ws$ catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
Base path: /home/michael/catkin_ws
Source space: /home/michael/catkin_ws/src
Build space: /home/michael/catkin_ws/build
Devel space: /home/michael/catkin_ws/devel
Install space: /home/michael/catkin_ws/install
Running command: "make cmake_check_build_system" in "/home/michael/catkin_ws/build"
Running command: "make -j6 -l6" in "/home/michael/catkin_ws/build"

[ 0%] Built target std_msgs_generate_messages_nodejs
[ 0%] Built target sensor_msgs_generate_messages_nodejs
[ 0%] Built target _realsense2_camera_generate_messages_check_deps_Metadata
[ 0%] Built target sensor_msgs_generate_messages_lisp
[ 0%] Built target _realsense2_camera_generate_messages_check_deps_DeviceInfo
[ 0%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[ 0%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[ 0%] Built target std_msgs_generate_messages_lisp
[ 0%] Built target sensor_msgs_generate_messages_eus
[ 0%] Built target std_msgs_generate_messages_eus
[ 0%] Built target sensor_msgs_generate_messages_cpp
[ 0%] Built target std_msgs_generate_messages_cpp
[ 0%] Built target std_msgs_generate_messages_py
[ 0%] Built target sensor_msgs_generate_messages_py
[ 0%] Built target _catkin_empty_exported_target
[ 0%] Built target tf_generate_messages_nodejs
[ 0%] Built target diagnostic_msgs_generate_messages_nodejs
[ 0%] Built target diagnostic_msgs_generate_messages_eus
[ 0%] Built target diagnostic_msgs_generate_messages_cpp
[ 0%] Built target dynamic_reconfigure_gencfg
[ 0%] Built target dynamic_reconfigure_generate_messages_cpp
[ 0%] Built target tf2_msgs_generate_messages_nodejs
[ 0%] Built target dynamic_reconfigure_generate_messages_lisp
[ 0%] Built target tf2_msgs_generate_messages_cpp
[ 0%] Built target tf2_msgs_generate_messages_lisp
[ 0%] Built target actionlib_generate_messages_py
[ 0%] Built target diagnostic_msgs_generate_messages_lisp
[ 0%] Built target rosgraph_msgs_generate_messages_nodejs
[ 0%] Built target actionlib_generate_messages_cpp
[ 0%] Built target actionlib_msgs_generate_messages_py
[ 0%] Built target roscpp_generate_messages_py
[ 0%] Built target roscpp_generate_messages_eus
[ 0%] Built target nav_msgs_generate_messages_cpp
[ 0%] Built target dynamic_reconfigure_generate_messages_py
[ 0%] Built target nav_msgs_generate_messages_eus
[ 0%] Built target geometry_msgs_generate_messages_eus
[ 0%] Built target actionlib_msgs_generate_messages_eus
[ 0%] Built target roscpp_generate_messages_cpp
[ 0%] Built target nav_msgs_generate_messages_lisp
[ 0%] Built target geometry_msgs_generate_messages_py
[ 0%] Built target tf2_msgs_generate_messages_eus
[ 0%] Built target roscpp_generate_messages_lisp
[ 0%] Built target actionlib_generate_messages_lisp
[ 0%] Built target geometry_msgs_generate_messages_cpp
[ 0%] Built target geometry_msgs_generate_messages_nodejs
[ 0%] Built target dynamic_reconfigure_generate_messages_eus
[ 0%] Built target std_srvs_generate_messages_cpp
[ 0%] Built target rosgraph_msgs_generate_messages_lisp
[ 0%] Built target tf_generate_messages_eus
[ 0%] Built target rosgraph_msgs_generate_messages_eus
[ 0%] Built target std_srvs_generate_messages_py
[ 0%] Built target roscpp_generate_messages_nodejs
[ 0%] Built target geometry_msgs_generate_messages_lisp
[ 0%] Built target tf_generate_messages_py
[ 0%] Built target tf_generate_messages_cpp
[ 0%] Built target actionlib_msgs_generate_messages_nodejs
[... 61 lines cut ...]
compilation terminated.
compilation terminated.
In file included from /usr/include/boost/range/iterator_range_core.hpp:38,
from /usr/include/boost/lexical_cast.hpp:30,
from /opt/ros/noetic/include/ros/transport_hints.h:34,
from /opt/ros/noetic/include/ros/subscribe_options.h:33,
from /opt/ros/noetic/include/ros/node_handle.h:42,
from /opt/ros/noetic/include/ros/ros.h:45,
from /opt/ros/noetic/include/image_transport/publisher.h:38,
from /opt/ros/noetic/include/image_transport/image_transport.h:38,
from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:8,
from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
from /home/michael/catkin_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/usr/include/boost/range/functions.hpp:18:10: fatal error: boost/range/begin.hpp: No such file or directory
18 | #include <boost/range/begin.hpp>
| ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:76: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:89: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o] Error 1
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:63: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1547: realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
Invoking "make -j6 -l6" failed

MartyG-RealSense, 2023-07-11:
The problems in the wrapper build seem to begin with this error:

`/usr/include/boost/range/functions.hpp:18:10: fatal error: boost/range/begin.hpp: No such file or directory`

Boost was removed from the current latest librealsense SDK version 2.54.1.  Which librealsense version do you currently have installed, please?  If it is 2.54.1 then I can see how the removal of Boost as a dependency in this SDK version might cause the **No such file or directory** error.  As you are using ROS Noetic and wrapper 2.3.2, the librealsense version must not be newer than **2.51.1**.

fitzzzzzzzzz, 2023-07-29:
I am also facing a same error and my versions are:
 2.50.0 of the librealsense SDK and 2.3.2 of the ROS wrapper
 have you found any solid fix for this.
 I am also running on jetson device with ubuntu 20.04
 
 please help @MartyG-RealSense

MartyG-RealSense, 2023-07-29:
Hi @fitzzzzzzzzz  The ROS1 wrapper (2.3.2) has not been updated since last year as development of it has ceased.  So the error would likely have been caused by something outside of the ROS wrapper as changes have not been made to the wrapper recently.  

What method did you use to install librealsense and the ROS wrapper on your Jetson, please?

fitzzzzzzzzz, 2023-07-29:
Hi @MartyG-RealSense , thank you so much for your quick response.

just a quick question, I am trying to reinstall again from the inputs you have shared above.

I am running this patch script from /scripts/patch-realsense-ubuntu-L4T.sh and I edited few lines like added 35.3.1 L4T version and jetpack as 5.1.1
I am getting a whole lot of errors, do you have any suggestion for that ? 

i want to build the librealsense with cuda for ros1 for further use, please help.

MartyG-RealSense, 2023-07-29:
If you are able to build librealsense from source code with CMake then using the **RSUSB backend** installation method avoids the need to use an edited patch script, as RSUSB bypasses the kernel and so it is not necessary to apply a kernel patch script when using RSUSB = true.

RSUSB is enabled by adding the flag **-DFORCE_RSUSB_BACKEND=TRUE** to the CMake build instruction.

If RSUSB does not work then it is also possible to use librealsense without applying a kernel patch, though the SDK will not provide hardware metadata information.

To enable CUDA support for Jetson in librealsense, you also need to include the flag **-DBUILD_WITH_CUDA=TRUE**

When CUDA support is enabled in librealsense, it is also automatically applied to the ROS wrapper.

fitzzzzzzzzz, 2023-07-31:
thanks a lot, its working for me. @MartyG-RealSense

MartyG-RealSense, 2023-07-31:
That's great to hear, @fitzzzzzzzzz - thanks very much for the update!

MartyG-RealSense, 2023-07-31:
@MichaelKerley  As the creator of this issue, do you require further assistance with it please?  Thanks!

fitzzzzzzzzz, 2023-07-31:
@MartyG-RealSense  I have one question though, what is the latest realsense ros-wrapper for jetpack 5.1. I want to use it with ros1 not with ros 2.
could you please help me out, becoz I am facing again above issue  again

MartyG-RealSense, 2023-07-31:
JetPack **5.0.2** is supported by librealsense 2.54.1 but not JetPack 5.1.  A couple of RealSense users have tried editing the L4T kernel patch script to add 5.1 support themselves but it didn't work out.  

So for 5.1 users I would recommend either not applying the kernel patch or build librealsense from source code with CMake with the **-DFORCE_DSUSB_BACKEND=TRUE** flag included in the CMake build instruction in order to bypass the kernel.

As you are using ROS1 and so librealsense **2.51.1** is the newest version that should be used with the ROS1 wrapper, not applying a kernel patch or using an RSUSB = true build of librealsense will work best for you if you have to use JetPack 5.1.

MartyG-RealSense, 2023-07-31:
You could try Ubuntu 18.04 and JetPack 4.6.

fitzzzzzzzzz, 2023-07-31:
okay thats the last option i hoped to try :(

sezarxray, 2024-10-07:
any comment from repo owner??????

fitzzzzzzzzz, 2024-12-18:
not yet

closed 2024-12-18 by MartyG-RealSense