realsenseai/librealsense #11979: rs.cpp:2704 - Couldn't refresh devices - hr returned: HResult 0x8007001f: " The device connected to the system is not functioning."

todayistipsy, 2023-07-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | PC ( dji manifold2) |
| SDK Version                            |  {2.53.1 }                          | 
| Language                            |  C++ (but now just in realsense viewer )                         | 
| Segment			|  Robot                  | 

### Issue Description
State : When I sametime use balser and realsense in a manifold2  (PC) ,I find my pc can recognize the device ,but in realsense viewer ,it tell me device can not be open like image underfollow
what is going on? is my d455‘s  problem?
How can i deal with it? plz help me,thanks

<Describe your issue / question / feature request / etc..>
[image]

[image]
[image]

MartyG-RealSense, 2023-07-09:
Hi @todayistipsy  Are you attempting to open the Basler USB3 camera in the RealSense Viewer tool, please?  

If the Basler is an RGB color camera then you should be able to access it in the Viewer by clicking the Add Source button at the top of the Viewer options panel and adding the Basler as a Platform Camera.

[image]

[image]

todayistipsy, 2023-07-09:
thanks for your answer。Actually  I try use realsense d455 and basler dart3840xx in one PC  for get different image stream,but when I use two usb of PC to connect them, I find I can not open the realsense separately。

todayistipsy, 2023-07-09:
it always tell me this message.
[image]

MartyG-RealSense, 2023-07-09:
A way to check for a conflict between the Basler camera and the RealSense D455 is to right-click on its Device Manager entry and select 'Disable device' from the menu.  You can easily re-enable the Basler after the test by right-clicking on it again and selecting 'Enable device'.

If the D455 appears in the Viewer after the Basler is disabled then there is likely to be a conflict with the Basler.

todayistipsy, 2023-07-09:
I try to find the botten——Platform Camera，but I can not find it.like this. and when I want use d455  separately.new problem is coming.... I can not adjust the parameter(the warning say) even I can not open the source of any camera
[image]

[image]

MartyG-RealSense, 2023-07-09:
Which firmware driver version is installed in your camera, please?  You can find the version by clicking on the 'Info' option.  Firmware 5.14.0.0 should be used with SDK version 2.53.1.

todayistipsy, 2023-07-09:
[image]
yes I use 05.14.00.00

MartyG-RealSense, 2023-07-09:
Thank you very much for the confirmation.

Does the same error occur if you download and run the file 'Intel.RealSense.Viewer.exe' from the Assets list at the link below?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

todayistipsy, 2023-07-09:
[image]
yes, it comes up this.

MartyG-RealSense, 2023-07-09:
Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice, please?

todayistipsy, 2023-07-09:
> Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice, please?

> Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice, please?

> Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice, please?
I use a usb unofficial cable

todayistipsy, 2023-07-09:
> Are you using the official 1 meter USB cable supplied with the camera or a USB cable of your own choice, please?

I use a unofficial usb cable.so maybe the problem is the cable?

MartyG-RealSense, 2023-07-09:
Using a high quality cable with RealSense cameras is important because of the high amount of data that the camera can transmit along the cable.  A cable designed for data transfer should be used.  An inexpensive cable for device charging or sync is not suitable.

MartyG-RealSense, 2023-07-15:
Hi @todayistipsy  Do you require further assistance with this case, please?  Thanks!

todayistipsy, 2023-07-16:
thanks! I think I find the problem.the Flex Interposer(As shown in the figure) between computer and camera is not work! (we use the module of d455 so not have the shell)
[image]

todayistipsy, 2023-07-16:
hoping can help other people to deal  the same problem

MartyG-RealSense, 2023-07-16:
Hi @todayistipsy  Thanks very much for your finding!  A flex interposer cable should work with any model of RealSense 400 Series cameras, though cables can of course develop problems.

closed 2023-07-31 by MartyG-RealSense