realsenseai/librealsense #11980: Realsense SDK v2.54.1 Rosbag file opening issue

NamalJayasuriya, 2023-07-09:
| | |
|---|---|
|**librealsense**|2.54.1 RELEASE|
|**OS**|Linux|

I updated realsense SDK to v2.54.1 and firmware of D415 camera to the recommended version. Now Realsense viewer crash with segmentation fault when I open previously recorded rosbag file using realsense viewer (with older version of SDK and Firmware). Recorded files after new update can be opened.

MartyG-RealSense, 2023-07-09:
Hi @NamalJayasuriya  I tested an 'objects submered under water' sample bag from 2017 that was recorded with a D415 and it played back in the 2.54.1 version of the RealSense Viewer with firmware version 5.15.0.2 without problems when the bag file was dragged and dropped into the center panel of the Viewer.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

Was the rosbag file that you are playing back recorded in an older version of the Viewer?  Rosbags that were recorded in ROS have incompatibility with playback in the RealSense SDK as it uses its own version of the rosbag format.

NamalJayasuriya, 2023-07-09:
Hi @MartyG-RealSense 

Thank you for your quick response.
I tried the same file that you tried and It loaded and played well. 
My rosbag files are recorded with latest firmware before this update.

MartyG-RealSense, 2023-07-09:
From around SDK 2.51.1 onwards, a single specific recommended firmware should be used for a specific SDK version or errors may occur.  The recommendations are:

5.13.0.50 for 2.51.1
5.14.0.0 for 2.53.1
5.15.0.2 for 2.54.1.  

Using a different firmware with those SDKs than the one recommended for it can result in errors.  So it is possible that a firmware conflict may have introduced problems in your bag recording.

NamalJayasuriya, 2023-07-09:
Thank you very much.
I found my old build from source SDK 2.50.0 and realsense-viewer tool their could open both bags recorded with firmware 5.14.0.0 and 5.15.02.

NamalJayasuriya, 2023-07-11:
As long as I have a way to quickly play my files, I do not worry. I'm using pyrealsense2 for further processing. 
However realsense viewer still recommended SW version: 2.54.1.5216, but when I go to the given link it shows version: 2.54.1.0 which I'm having now, could you please send me the link for new version?

MartyG-RealSense, 2023-07-11:
You don't need to be concerned about the minor number on the end of the version name so long as it is version 2.54.

MartyG-RealSense, 2023-07-24:
Case closed due to no further comments received.

closed 2023-07-24 by MartyG-RealSense

NamalJayasuriya, 2023-09-25:
@MartyG-RealSense Thank you very much!