realsenseai/librealsense #11982: D405 cand view RGB on realsense-viewer

DaveBGld, 2023-07-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Ubuntu20.04 | 
| Kernel Version (Linux Only)    |  5.15.0-76-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  latest built from code2.54.1.0  | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
When running realsense-viewer only the depth channes is available. See the screenshot:

[image]

Thanks for the help!

MartyG-RealSense, 2023-07-09:
Hi @DaveBGld  On the D405 model, the RGB options are under the **Stereo Module** section of the Viewer's options side-panel when the list of options is expanded open by clicking on the small arrow beside Stereo Module.  

This is because D405 does not have a separate RGB sensor and instead obtains RGB from the depth sensor and processes it through an Image Signal Processor (ISP) chip inside the camera to generate the final RGB image.

[image]

DaveBGld, 2023-07-10:
@MartyG-RealSense Thanks for the explanation!!

I selected color, and still don't get an RGB image... Even more so, I see there are many "black" areas where there is no depth measurement?

Is there a link to the meanings and intellisense2 code settings for the left panel?

[image]

Many thanks!

MartyG-RealSense, 2023-07-18:
Hi @DaveBGld  Do you require further assistance with this case, please?  Thanks!

closed 2023-07-24 by MartyG-RealSense

AndreV84, 2024-11-18:
> f options is expanded open by clicking on the small arrow beside

@MartyG-RealSense seems realsense viewer [ installed 2.55.1 from focal apt repo at pi5 bookworm OS] won't show color stream regardless two checkbox checked for depth and color before pressing the toggle button; it would only show the depth stream while the color would be empty; is there other way to confirm color stream works on the deployment unit?
However, the suggested solution seems may be to install libuvc script thing/ or it doesn't address the empty color stream specifically? 
Thanks
AV

AndreV84, 2024-11-19:
@MartyG-RealSense Thank you for your prompt reply!
in the example [#12419](https://github.com/IntelRealSense/librealsense/issues/12419)  
user utilizes  a different specie ; `Orange` Pi
It is different from `Raspberry` Pi  models;
could you share download url for source code of 2.55.1.6486, please ?
I did just default git clone on the main repo and got realsense-viewe message that says to update software from 2.55.1 to 2.55.1.6486; I tried to locate such specific version but yet haven't found  download url:)
Thank you very much !
AV

MartyG-RealSense, 2024-11-20:
You can download the source code from the 'Assets' file list at the bottom of the page at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.55.1

[image]

However, in practice you can ignore the message about updating to a long-numbered version as there should be virtually no difference between the short-numbered and long-numbered versions (such as 2.55.1 and 2.55.1.6486).