realsenseai/librealsense #11986: Programatic camera reset

DaveBGld, 2023-07-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Ubuntu20.04 | 
| Kernel Version (Linux Only)    |  5.15.0-76-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  latest built from code2.54.1.0  | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot}                          | 

We currently need to use the same camera with very different settings, and the only way to achieve this is disconnecting and reconnecting the camera. Not good for production.

Is there a way to do a camera reset  equivalent to that unplug/plug?

Many thanks!

David

MartyG-RealSense, 2023-07-10:
Hi @DaveBGld  You do not need to unplug the camera to apply new settings.  You can simply stop the pipeline with an instruction such as **pipe.stop()**, and then place a new set of cfg instructions after the stop instruction or load a **json** camera configuration file.  When the pipeline is started again with a **pipe.start()** instruction then the new settings will be applied.

If you need to reset the camera without an unplug-replug then you can do so with the **hardware_reset()** script instruction, like the C++ example at https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

You could also reset the camera by resetting the entire USB port with an Ubuntu **bash script** like the example at https://github.com/IntelRealSense/librealsense/issues/8393

An advantage of using a bash script is that the camera does not need to be detected in order for the reset to be initiated.

DaveBGld, 2023-07-20:
> f you need to reset the camera without an unplug-replug then you can do so with the **hardware_reset()** script instruction, like the C++ example at [#2161 (comment)](https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150)

After **hardware_reset()** and restarting the pipe and re-configuring, I get timeout no frames received.

DaveBGld, 2023-07-20:
> What happens with the first suggestion from [#11986 (comment)](https://github.com/IntelRealSense/librealsense/issues/11986#issuecomment-1629750129) of stopping the pipeline, applying a new set of cfg instructions and then starting the pipeline again?

That's the start of the problem. After setting manual exposure, only unplugging the camera will allow a new configuration

MartyG-RealSense, 2023-07-20:
The main settings that require a pipeline stop and then start to change are resolution and FPS.  With most other settings, including manal exposure, you should be able to change their value with an instruction such as **sensor.set_option(RS2_OPTION_EXPOSURE, 33000);** whilst the program is running without stopping the pipeline.

On the D405 camera model, you need to access the depth sensor and set a depth exposure value to change the RGB exposure, as D405 does not have a separate RGB sensor and so does not have a separate RGB exposure setting.  Increasing depth exposure will brighten the RGB image and decreasing depth exposure will darken the RGB image.

closed 2023-08-02 by MartyG-RealSense