realsenseai/librealsense #11987: D435i IMU recording at inconsistent rate

ryanalex98, 2023-07-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i {D400 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-76-generic      | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.1.0 (downloaded via command line, not built from source)    | 
| Language                            |  C++    | 
| Segment			|  Robot             | 

### Issue Description

Hi there, I am currently using a D435i camera and attempting to record some IMU data. However, when I record data, I am noticing a very inconsistent rate at which the readings are recorded. I have reviewed several issues in this forum to try and alleviate the problem across different product models but nothing seems to work for my case, hence why I am opening my own issue. I will do my best to go through what I've tried thus far.

#### First attempt: Gyro 400Hz, Accel 250Hz

USB 3.2 cable, port.

**Launch file**:
[imu_recording.txt](https://github.com/IntelRealSense/librealsense/files/12008973/imu_recording.txt)

**Command line output**
```
ryanalexander@ryanalexander-ThinkPad-P51:~/repos/custom/DataRecording$ roslaunch imu_recording imu_recording.launch 
... logging to /home/ryanalexander/.ros/log/50baf114-1f71-11ee-b0ee-1b5bbb1d9f49/roslaunch-ryanalexander-ThinkPad-P51-12884.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ryanalexander-ThinkPad-P51:34191/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 5
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 5
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 720
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 1280
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: False
[... 127 lines cut ...]
[ INFO] [1689028215.789152209]: RealSense Node Is Up!
 10/07 23:30:51,933 **WARNING [140112033740544] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11**
```

**Rosbag investigation**:
As you can see, the timestamps are consistently inconsistent, with drops occurring around every second.
[image]

#### Second attempt with camera enabled

When I enable the infra1 and infra2 cameras at 30Hz (640x480), I get the following output:
[image]
On the surface this output looks correct, but it should be operating at 400Hz, or every 0.0025 seconds. Instead, its alternating between producing a single message and then a couplet of two messages roughly every 0.0037 seconds. So on average, it produces three messages every 0.0075 seconds which lines up with 400Hz - the distribution of the messages is the problem.

#### Issues I referenced while attempting debug:
- https://github.com/IntelRealSense/realsense-ros/issues/1756
- https://github.com/IntelRealSense/realsense-ros/issues/1663
- https://github.com/IntelRealSense/realsense-ros/issues/1469
- https://github.com/IntelRealSense/realsense-ros/issues/1607#issuecomment-760128960
- https://github.com/IntelRealSense/realsense-ros/issues/2127
- https://github.com/IntelRealSense/librealsense/issues/10427

I tried decreasing the rates to 200Hz (Gyro) and 63Hz (Accel), but this didn't help. I also toggled global_time_enabled:=false (and true), neither of which solved the issue.

Please let me know if there is any more information I can provide in resolving this case. Thank you very much!

MartyG-RealSense, 2023-07-11:
Hi @ryanalex98  Setting **unite_imu_method** to 'copy' instead of 'linear_interpolation' may provide more stable IMU data.  This can be done by including the command **unite_imu_method:=copy** in your roslaunch instruction.

As well as setting global time to false, you could also try setting **enable_sync** to true with the roslaunch command **enable_sync:=true** if you have not done so already.  When true, it gathers the closest frames of different sensors to be sent with the same timetag.

ryanalex98, 2023-07-12:
Thanks for the quick reply @MartyG-RealSense.

I updated the launch file with the unite_imu_method set to copy and launched with the following args:
```
roslaunch imu_recording imu_recording.launch global_time_enabled:=false enable_sync:=true
```

Still getting curious output behavior:
[image]

I think I have the versions aligned, I'm using sdk v2.54.1.0 and firmware v5.15.0.2, which I believe are the most recent release pair?

Thanks for your help.

MartyG-RealSense, 2023-07-13:
Thanks very much for the update.  Yes, SDK 2.54.1 and firmware 5.15.0.2 are the correct pairing.

https://github.com/IntelRealSense/realsense-ros/issues/898 is an interesting discussion about IMU stability which led to the original creation of the 'copy' mode of unite_imu_method to provide additional IMU data stability.  Within that discussion, a RealSense user reported at https://github.com/IntelRealSense/realsense-ros/issues/898#issuecomment-790278100 that they still had some erratic data after 'copy' mode was implemented and they shared their chart.

[image]

closed 2023-07-27 by MartyG-RealSense