realsenseai/librealsense #11994: Issue building librealsense from source using colcon

martinakos, 2023-07-13:
I want to build the librealsense from source in ros2 humble using colcon. I do the following

I first install the dependencies:
`sudo apt install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev libusb-1.0-0-dev`

I clone the librealsense2 repo (master) in the src folder of my ros2 workspace (/home/user/realsense2), and run the following command from that workspace:
`colcon build --packages-up-to librealsense2  --cmake-args -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=True`

I need the RSUSB backend, that's why I'm building it from source. It builds fine but when I try to source the setup.bash I get this error:

```
. install/setup.bash 
not found: "/home/user/realsense2/install/librealsense2/share/librealsense2/local_setup.bash"
```

I know this is not the realsense-ros package, so maybe I shouldn't be surprised there is no local_setup.bash
but I'm building a number of ros2 packages (including the realsense-ros package) and for deployment it's convenient to build everything (including librealsense2) with colcon. 
Also, notice that if a ros2 package depends on librealsense2, it would try to source that ./install/librealsense2/share/librealsense2/local_setup.bash file when sourcing it's own setup.bash which in turn will fail.

Does anybody have a suggestion to avoid the above error?

MartyG-RealSense, 2023-07-13:
Hi @martinakos  In the RealSense ROS1 wrapper at least, the package installation (building librealsense and the wrapper together at the same time) was based on the RSUSB Backend.  I do not know if the ROS2 packages are also based on RSUSB, but it may be worth trying the building of the SDK and wrapper together with the command below:
 
`sudo apt install ros-<ROS_DISTRO>-librealsense2*`

A package installation of librealsense via the ROS wrapper will not include the RealSense Viewer tool, so librealsense and the wrapper should be built separately if you need the Viewer.

I note your preference for a colcon build though because you are building multiple ROS2 packages.

If you are planning on installing the same librealsense / ROS2 setup on a number of computers then instead of just one, you could consider installing everything the 'long way' on one computer, and then creating an **iso** backup image file of the entire computer and writing that iso image to a clean drive on another computer to create an identical copy of the original computer.

MartyG-RealSense, 2023-07-14:
I was unable to find a previous example of librealsense and the ROS wrapper being built together from source using colcon build, unfortunately.

Instructions for building librealsense from source are here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

The CMakeLists.txt file for building librealsense from source is in the root folder of the librealsense source code:

https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt

It is usually accessed by creating a directory called **build** in the root folder and then navigating to it and inputting a CMake build instruction, such as `cmake ../ -DFORCE_RSUSB_BACKEND=true`

closed 2023-07-21 by martinakos