realsenseai/librealsense #11999: Reading a rosbag message with C++

JuanMiguelGV, 2023-07-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.34.0                           | 
| Language                            |  C++                          | 
| Segment			|  Robot                      | 

### Issue Description
Hello:

I'm finishing up a big project in C++ and the only thing left is to acquire the frames per second at which the different streams were recorded. I'm trying to get it directly using rosbag, but the way in which this message is recorded and the format for it has made it quite complicated due to dependencies and other compiler errors I haven't been able to overcome.

In python, the solution is very simple, I just used the following to get the fps of both sensors I'm interested in:

```
import rosbag
bag = rosbag.Bag(path)
fps_s0 = next(bag.read_messages(topics=['/device_0/sensor_0/Depth_0/info'])).message.fps
fps_s1 = next(bag.read_messages(topics=['/device_0/sensor_1/Color_0/info'])).message.fps
```

But my project is in C++, and it's been a nightmare to get this simple message of the bag file.
`print_helpers.h` and `rosbag_content.h` somehow are not found by my project, even after adding the right dependencies and even copying the `rs-rosbag-inspector` project as a template.

After scratching the library, I came up with this, but does not work still. I have currently a lot of LNK2019 errors referring to the .obj file with the name of my project, even tough I added all additional include directories and linker input of the `rs-rosbag-inspector`

```
#include "rosbag/bag.h"
#include "rosbag/view.h"
#include "../../third-party/realsense-file/rosbag/msgs/realsense_msgs/StreamInfo.h"

std::string depth_fps;
std::string color_fps;

template <typename ROS_TYPE>
inline typename ROS_TYPE::ConstPtr try_instantiate(const rosbag::MessageInstance& m)
{
    if (m.isType<ROS_TYPE>())
        return m.instantiate<ROS_TYPE>();
    return nullptr;
}

void main()
{
    rosbag::Bag bag;
    bag.open("Path to bag file", rosbag::bagmode::Read);

    std::vector<std::string> topics;
    topics.push_back(std::string("/device_0/sensor_0/Depth_0/info"));
    topics.push_back(std::string("/device_0/sensor_1/Color_0/info"));

    rosbag::View view(bag, rosbag::TopicQuery(topics));

    for (rosbag::MessageInstance const m : view)
    {
        if (auto data = try_instantiate<realsense_msgs::StreamInfo>(m))
        {
            auto stream_info = data;
            if (m.getTopic() == ("/device_0/sensor_0/Depth_0/info"))
            {
                depth_fps = stream_info->fps;
            }
            else
            {
                color_fps = stream_info->fps;
            }
        }

    }
}
```

Any help would be very appreciated, I wasn't expecting getting the messages from the bag file to be so complicated.

MartyG-RealSense, 2023-07-14:
Hi @JuanMiguelGV  If you are compiling your C++ project in Visual Studio on Windows and are experiencing LNK2019 errors then the best way to resolve these may be to set up the RealSense linkages in your project using three **.props** property sheet files provided by the RealSense SDK.  A RealSense user shared a starter tutorial project for doing so at the link below.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

JuanMiguelGV, 2023-07-14:
Hi @MartyG-RealSense, thanks for your response. I added them to the project after it was created, I tried again just to double check, but there was no effect. Any ideas on how to retrieve this info I need? I rather get it with rosbag instead of counting the exact number of frames and dividing by the time.

Here is a [screenshot ](https://imgur.com/EoGWKvY) of the errors I'm getting

MartyG-RealSense, 2023-07-15:
Counting frames and dividing by time is an inaccurate way to calculate FPS.  It con be obtained from a camera metadata parameter called **ACTUAL_FPS**.  Metadata should be recorded into a bag along with the streams.

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387 has an example of a C++ script that prints the value of a metadata parameter with an **std::cout** command.

```
 frameset = pipe.wait_for_frames();
 std::cout << "meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;
```

To access the REAL_FPS metadata parameter, you could change ACTUAL_EXPOSURE to ACTUAL_FPS.

`frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_FPS) << std::endl;`

JuanMiguelGV, 2023-07-19:
Thanks for the alternative, it was quite useful!

I still went over it again, and finally was able to implement my code, being very cautious with the project configuration. For further reference, if anyone wants to use the `print_helpers.h`, `rosbag_content.h` or the `files_container.h` to read the bag files **without** a user interface as the one from `rs-rosbag-inspector`:

Add these to Additional Include Directories:

```
\librealsense-X.XX.X\tools\rosbag-inspector
\librealsense-X.XX.X\third-party\realsense-file\rosbag\console_bridge\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\cpp_common\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\rosbag_storage\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\roscpp_serialization\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\rostime\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\roscpp_traits\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\roslz4\include
\librealsense-X.XX.X\third-party\realsense-file\rosbag\msgs
\librealsense-X.XX.X\third-party\realsense-file\boost
\librealsense-X.XX.X\third-party\realsense-file\lz4\lib
```

Add this library to the Linker → Input → Additional dependencies:

`librealsense-X.XX.X\build\third-party\realsense-file\Release\realsense-file.lib` (for debug: `\Debug\realsense-filed.lib`)

And finally, make sure your C/C++ → Code generation → Runtime Library is set in /MT for release or /MTd for debug. (Using /MD or MDd won't allow the project to compile).

I also had to make some changes in a couple of the SDK files, as my project uses the C++17 standard, so I strongly advise staying with the C++14 when using the SDK.

closed 2023-07-19 by JuanMiguelGV

MartyG-RealSense, 2023-07-19:
It's great to hear that you were successful, @JuanMiguelGV - thanks very much for the update and for sharing your detailed solution!