realsenseai/librealsense #12000: D435 can't detect by Nvidia Jstson AGX Xavier

ericsuntj, 2023-07-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   (Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  { legacy / 2.54/2.45 }                          | 
| Language                            |  {C++/ }                          | 
| Segment			|  {others }                          | 

### Issue Description
I use a USB3.0 to connect the D435 with NVIDIA Jetson AGX Xavier. But the AGX can't detect the D435 completely. I use the `lsusb` but it still can't list the D435 and the `realsense-viewer` don't work as well.

MartyG-RealSense, 2023-07-16:
Hi @ericsuntj  Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

ericsuntj, 2023-07-18:
> 

Thanks for reply! Yes, I'm using the official USB cable.

closed 2023-08-02 by MartyG-RealSense