realsenseai/librealsense #12004: advanced_mode_example failed

tonyzzzzz, 2023-07-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400} | 5.15.0.2
| Operating System & Version |   {Win (10) 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {Matlab }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
We tried to configure the RGB camera's exposure via advanced mode APIs. However, it looks like librealsense_mex won't let us set the parameter. Even the advanced_mode_example.m came along with the SDK failed (see error messages below). Can someone please shed some lights here?

Matlab Code (copied from advanced_mode_example.m)
==========================================
json_string = adv.serialize_json();
json = jsondecode(json_string);
%line below generate error, hence we comment it out.
%json.controls_color_gain = string(str2double(json.controls_color_gain) - 4);
json_string = jsonencode(json);
% matlab replaces - with _ to create valid structure field names, we now need to switch them back
json_string = strrep(json_string, '_', '-');
% Upload new json
adv.load_json(json_string);

Matlab Error Message:
=============================
error using librealsense_mex
preset file is corrupt, cannot validate schema
error realsense.advanced_mode/load_json (line 30)
            realsense.librealsense_mex('rs400::advanced_mode', 'load_json', this.objectHandle, json_content);

MartyG-RealSense, 2023-07-17:
Hi @tonyzzzzz  How did you create your json camera configuration file, please?  Did you export it from the RealSense Viewer tool using the json export button near the top of the Viewer's options side-panel?  This method of defining the value of settings in the Viewer and then exporting them is the easiest way to generate a json that does not have faults in it.

[image]

MartyG-RealSense, 2023-07-17:
There has not been a past report of this error at the time of writing this.  If another RealSense MATLAB wrapper user reports it then I will highlight it to my Intel RealSense colleagues.

In the meantime, a code snippet at https://github.com/IntelRealSense/librealsense/issues/5236#issuecomment-594555256 for loading a json file in the MATLAB wrapper may be helpful.

tonyzzzzz, 2023-07-17:
thanks. I think we can close this issue now. 

btw - It is obvious for me that the Json format assumed in advanced_mode_example.m is different from the json format exported from Realsenseviewer.

MartyG-RealSense, 2023-07-18:
Thanks very much @tonyzzzzz for the update.  As you suggest, I will close the issue.  Thanks again!

closed 2023-07-18 by MartyG-RealSense