realsenseai/librealsense #12005: Camera stuck in  "Resetting device..."

martinakos, 2023-07-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |5.15.0.2| 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 5.15.0-72-generic                                        | 
| Platform                                 | PC.  |
| SDK Version                            |  v2.54.1                          | 
| ROS driver                            |  RealSense ROS v4.54.1                         | 

I rebuild the librealsense using the RSUSB backend. It works well. But now when I use it with the ROS driver and I have initial_reset: true the driver stays stuck in the  "Resetting device..." message forever. If I set the initial_reset: false it starts normally.  It didn't use to do this with the V4L backend. 

For the moment I'm setting initial_reset: false but any idea why this may happen?

MartyG-RealSense, 2023-07-17:
Hi @martinakos  I have seen a RealSense ROS case like this before. I believe that all that you can do is not use initial_reset.  The reset is not a compulsory option anyway and only needs to be used if there is a problem with the launch that a reset helps to resolve.

martinakos, 2023-07-17:
Ok, thanks. I'll keep the initial_reset: false then.