realsenseai/librealsense #12006: failed to set exposure duration from Realsense viewer

tonyzzzzz, 2023-07-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   {Win 10)| 
| Platform                                 | PC
| SDK Version                            |  2.0

### Issue Description
tried to set up the exposure duration to 33 ms from the Realsense viewer. From the [link] (https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance?_ga=2.211252373.1832360664.1689613205-1370688860.1688918478#section-make-sure-image-is-properly-exposed), the unit is micro seconds, hence I suppose that I can set the exposure duration to 33000 from the Realsense viewer (once Auto-Exp is disabled). However, I got the error saying "33000 is out of bounds [1,10000]". I guess perhaps the unit used in Realsense viewer is not microsecond?

[image]

MartyG-RealSense, 2023-07-18:
Hi @tonyzzzzz  Depth and RGB exposure use different scales in their values.  Depth exposure supports large values such as 33000 and RGB exposure has smaller values such as 156.

tonyzzzzz, 2023-07-18:
Thanks, but what is the unit for the RGB exposure duration? Still micro sec?

Best regards
Tony

MartyG-RealSense, 2023-07-18:
The reason why only certain exposure values are permitted on Windows is explained by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/5493#issuecomment-567857540

The depth exposure unit is microseconds, as confirmed in point 2 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#make-sure-image-is-properly-exposed

There is a reference in the SDK's operating code that states the units of the RGB sensor's exposure to also be in microseconds.

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/ds5/ds5-color.cpp#L135

tonyzzzzz, 2023-07-18:
Got it. #5493 explained it very well, thanks.

Per #5493, I think the unit for windows exposure is 0.1 ms instead of microsecond.

I think we can close this issue.:)

Best regards
Tony

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#make-sure-image-is-properly-exposed

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/src/ds5/ds5-color.cpp#L135

MartyG-RealSense, 2023-07-18:
You are very welcome.  As you suggest, I will close the issue as you are satisfied.  Thanks very much!

closed 2023-07-18 by MartyG-RealSense