realsenseai/librealsense #12009: 在VS2022上运行rs-hello-realsense示例代码未在控制台输出结果

NET-Y, 2023-07-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   |5.13.0.55 | 
| Operating System & Version |   win10  | 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC |
| SDK Version                           | SDK  2.54.1                  | 
| Language                              |  C++ }                          | 
| Segment			        |  {Robot }                          | 

### Issue Description
在VS上运行Examples for Intel RealSense SDK 2.0中的示例代码时只能显示摄像头画面窗口，而控制台中没有任何结果输出
猜测可能是环境配置出现问题，能用什么方法确定环境配置是否正确
[image]

这是我们使用的示例代码
[rs-hello-realsense.zip](https://github.com/IntelRealSense/librealsense/files/12077490/rs-hello-realsense.zip)

MartyG-RealSense, 2023-07-18:
Translation: Running the rs-hello-realsense sample code on VS2022 does not output the result on the console

When running the sample code in Examples for Intel RealSense SDK 2.0 on VS, only the camera screen window can be displayed, but there is no output in the console.  It is guessed that there may be a problem with the environment configuration. 
 What method can be used to determine whether the environment configuration is correct?

*********

Hi @NET-Y  The RealSense SDK examples such as rs-hello-realsense are compiled in a different way to how programs created by RealSense users are built, so if the code of the examples is copied over into your own project then it will usually not work because some of the linkages to the librealsense library will be missing.

One way to build rs-hello-realsense as a standalone program is to build it with CMake using the tutorial project at the link below.  It uses the same script as rs-hello-realsense but has a more complex CMakeLists.txt build file that contains the missing librealsense linkages.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

If you are using VS2022 and Windows though then the best approach to setting up the linkages to librealsense in your project is likely to be to use the three VS **.props** property sheet files provided by the SDK.  There is a tutorial for doing this here:

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

*************

您好@NET-Y RealSense SDK 示例（例如 rs-hello-realsense）的编译方式与构建 RealSense 用户创建的程序的方式不同，因此，如果将示例的代码复制到您自己的项目中，那么通常会 不起作用，因为与 librealsense 库的一些链接将丢失。

将 rs-hello-realsense 构建为独立程序的一种方法是使用下面链接中的教程项目通过 CMake 进行构建。 它使用与 rs-hello-realsense 相同的脚本，但具有更复杂的 CMakeLists.txt 构建文件，其中包含缺少的 librealsense 链接。

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

如果您使用的是 VS2022 和 Windows，那么在项目中设置 librealsense 链接的最佳方法可能是使用 SDK 提供的三个 VS **.props** 属性表文件。 这里有一个执行此操作的教程：

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

MartyG-RealSense, 2023-07-27:
Do you require further assistance with this case, please?  Thanks!

请问您对此案需要进一步的帮助吗？ 谢谢！

MartyG-RealSense, 2023-08-02:
Case closed due to no further comments received.

由于没有收到进一步的评论，案件已结。