realsenseai/librealsense #12012: Jetpack 5.1.1 ( L4T 35.3.1) support ?

Arthav24, 2023-07-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415/435/455/435i | 
| Firmware Version                   | 05.15.00.02 | 
| Operating System & Version |   Linux (Ubuntu 20.05)  | 
| Kernel Version (Linux Only)    |  5.10   | 
| Platform                                 |  NVIDIA Jetson Orin AGX/Nano  |
| SDK Version                            |  v2.54.1 | 
| Language                            |  C/CPP/python ROS(noetic)/ROS2(humble)           | 
| Segment			|  Robot                          | 

any ETA for Jetpack 5.1.1 ( L4T 35.3.1) support ? I tired running ./scripts/patch-realsense-ubuntu-L4T.sh  but fails. I intend to run the configuration in production so kinda avoiding DFORCE_RSUSB_BACKEND. More inclined to Native V4L.

MartyG-RealSense, 2023-07-18:
Hi @Arthav24  I do not have an ETA for official JetPack 5.1.1 support at the time of writing this (5.0.2 is officially supported in SDK 2.54.1).  A couple of RealSense users at https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762 have been able to successfully use the SDK with Jetson Orin without RSUSB by not applying the kernel patch script though.

Arthav24, 2023-07-18:
I tried DFORCE_RSUSB_BACKEND and the docker image from Dockerfile ,but I could perceive noticeable lag in depth and rgb streams in just launching through librelasense_ros. Thus started to explore kernel patching.
Thanks for quick help on it.

MartyG-RealSense, 2023-07-18:
You are very welcome.  Please do let me know if you require further assistance.  Thanks!

Arthav24, 2023-07-21:
Hi @MartyG-RealSense I was able to patch my jetpack 5.1.1 kernel 5.10 with slight modifications in script. I figured out that JP5.0.2 and 5.1.1 uses same kernel so things should work as expected.
But while installation i encountered ```Reversed (or previously applied) patch detected!  Assume -R? [n]``` error to which I proceeded with Y. According to documentation the patch detected that the change contained in the patch seems to have
already been made. so does JP5.1.1 doesn't needs to be patched explicitly ?

Also let me know if I should attached modified files for JP5.1.1 support.

MartyG-RealSense, 2023-07-21:
The operating principles and editing of kernel patch scripts is not a subject that I have knowledge of, unfortunately.  Does using your edited version of the patch script appear to work?

Arthav24, 2023-07-21:
Does using your edited version of the patch script appear to work? YES

I am in process to evaluate the gain of using with CUDA. else I could see poincloud and rgb on rviz.

MartyG-RealSense, 2023-07-21:
Okay, it's great to hear that your edit is working for you.

If you have not added librealsense's CUDA support already then you can do so either by installing librealsense from the Jetson version of the debian packages (which automatically include CUDA support) or by building the SDK from source code with CMake and including the flag **-DBUILD_WITH_CUDA=ON** in the CMake build instruction.  When CUDA support is enabled in the SDK, pointclouds and alignment will automatically benefit from acceleration by offloading processing work from the CPU onto the Jetson's Nvidia graphics GPU.

closed 2023-07-22 by Arthav24

jzx-gooner, 2023-09-01:
I build from source use DFORCE_RSUSB_BACKEND on jetson orin with jetpack 5.1.1
use realsense_ros publish the image and depth and it works fine
However when i run plane fitting by myself code and  i get different results from x86 and jetson .it is very strage,it means that jetson version publish the wrong depth??

MartyG-RealSense, 2023-09-01:
@jzx-gooner  Have you included the flag **-DBUILD_WITH_CUDA=ON** in your CMake instruction for building librealsense from source code so that the RealSense SDK's CUDA graphics acceleration support for Jetson is enabled?  Does having CUDA enabled make a difference to the published depth?

jzx-gooner, 2023-09-02:
@MartyG-RealSense  thank you ! yes .i use -DBUILD_WITH_CUDA=ON

MartyG-RealSense, 2023-09-03:
@jzx-gooner  There have been a small number of reported past cases where depth was reported correctly on an x86 PC but incorrectly on a Jetson.  One RealSense user's solution was to set -DBUILD_WITH_CUDA to False.  The CUDA support in librealsense mainly accelerates depth-to-color alignment and pointclouds, so if your program uses neither of these then setting CUDA to FALSE instead of ON may not have a negative effect to the program's performance.