realsenseai/librealsense #12014: On-Chip calibration fails with "hwmon command 0x80( 8 2 0 8 ) failed (response -7= HW not ready)"

stwirth, 2023-07-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.54.1                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description
I'm trying to get the on chip calibration going in a script.

I can run it fine from the realsense viewer:
[image]

But when I try to run it from a script, e.g. [depth_auto_calibration_example.py](https://github.com/IntelRealSense/librealsense/blob/8ffb17b027e100c2a14fa21f01f97a1921ec1e1b/wrappers/python/examples/depth_auto_calibration_example.py) I get this error:

```
$ python3 depth_auto_calibration_example.py 
Starting UCAL...
Starting On-Chip calibration...
	Speed:	medium
	Scan:	intrinsic
Traceback (most recent call last):
  File "depth_auto_calibration_example.py", line 285, in <module>
    main()
  File "depth_auto_calibration_example.py", line 87, in main
    run_on_chip_calibration(args.onchip_speed, args.onchip_scan)
  File "depth_auto_calibration_example.py", line 121, in run_on_chip_calibration
    table, health = adev.run_on_chip_calibration(args, progress_callback, 30000)
RuntimeError: hwmon command 0x80( 8 2 0 8 ) failed (response -7= HW not ready)
```

What am I doing wrong?

MartyG-RealSense, 2023-07-19:
Hi @stwirth  When calibration errors such as this one occur then the calibration may work after several restarts of the program.  There is usually not a known alternative solution to performing multiple restarts, unfortunately.  Having said that, a RealSense user at https://github.com/IntelRealSense/librealsense/issues/7087#issuecomment-730188812 said that they were able to create a Python script that worked.

stwirth, 2023-07-19:
Thanks @MartyG-RealSense for the reference, I had not enabled the stream in my C++ script. Following the code sample from https://github.com/IntelRealSense/librealsense/issues/7087#issuecomment-1435039708 with an added initial reset it started working for me.

closed 2023-07-19 by stwirth

MartyG-RealSense, 2023-07-19:
I'm pleased to hear that you achieved a working solution.  Thanks very much for the update!

gingerZone27, 2025-04-14:
In function `def run_on_chip_calibration(speed, scan)` there are two lines:
```python
adev = dev.as_auto_calibrated_device()
table, health = adev.run_on_chip_calibration(args, progress_callback, 30000)
```

I added one line between these two and it worked:
```python
adev = dev.as_auto_calibrated_device()
time.sleep(1.0)
table, health = adev.run_on_chip_calibration(args, progress_callback, 30000)
```

It seems that `adev = dev.as_auto_calibrated_device()` needs some time to configure the hardware.
Otherwise, the error occurs: `hwmon command 0x80( 8 2 0 8 ) failed (response -7= HW not ready)`, showing that the hardware is not ready yet.

Not sure if this is the final solution but it's worth a trial.

MartyG-RealSense, 2025-04-14:
Thanks so much @gingerZone27 for sharing your experience with addressing this problem!