realsenseai/librealsense #12015: running rs-kinfu on ubuntu 20.4

Nimaro76, 2023-07-18:
Hello, 
I tried running rs-kinfu.cpp example both on Windows and Ubuntu. In both cases, I had problems regarding the openGL dependencies in the code. These errors occur despite the fact that I have installed and tested the openGL and glfw3 on my laptop. I also tested that opencv is installed correctly as well. In the attachment, I send you the errors that I get in Ubuntu 20.4 when running rs-kinfu. cpp. I would appreciate it if you guided me. if rs-kinfu.cpp source code is not working, could you please provide me with alternatives? I want to do 3D scene reconstruction, similar to what you did with SLAM DEMO in ROS ( https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i ) However, I don't want the reconstructed 3D scene visualization to be in ros. thank you for your support.
errors on ubuntu : 
/usr/bin/ld: /tmp/ccrtw961.o: in function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
rs-kinfu.cpp:(.text+0xe61): undefined reference to `glLoadIdentity'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xe6b): undefined reference to `glPushAttrib'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xe90): undefined reference to `glClearColor'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xe9a): undefined reference to `glClear'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xea4): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xea9): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xee4): undefined reference to `gluPerspective'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xeee): undefined reference to `glMatrixMode'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xef3): undefined reference to `glPushMatrix'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xf3a): undefined reference to `gluLookAt'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xf6f): undefined reference to `glTranslatef'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xf95): undefined reference to `glRotated'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xfba): undefined reference to `glRotated'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xfcf): undefined reference to `glTranslatef'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xfe5): undefined reference to `glPointSize'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xfef): undefined reference to `glEnable'
/usr/bin/ld: rs-kinfu.cpp:(.text+0xff9): undefined reference to `glEnable'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x1015): undefined reference to `glBindTexture'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x105f): undefined reference to `glTexParameterfv'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x1073): undefined reference to `glTexParameteri'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x1087): undefined reference to `glTexParameteri'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x1091): undefined reference to `glBegin'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x1139): undefined reference to `glVertex3fv'
/usr/bin/ld: rs-kinfu.cpp:(.text+0x115d): undefined reference to `glTexCoord2fv'

Nimaro76, 2023-07-19:
on Windows, I build from source by cloning the Github repository to a certain directory. I will try the ubuntu installation one more time and get back to you. thanks

Nimaro76, 2023-07-19:
regarding the mentioned 3D reconstruction suggestions, What I need is a C++ or python code for doing the spatial mapping like what SLAM  does in ros or like kimera-VIO but both of these are in ROS. I hope rs-kinfu runs as this is probably the ideal choice

MartyG-RealSense, 2023-07-19:
My understanding is that **Kimera-VIO** is not dependent on ROS but users have the option to interface with it via the Kimera-VIO-ROS wrapper.

https://github.com/MIT-SPARK/Kimera-VIO

https://github.com/MIT-SPARK/Kimera-VIO#ii-using-ros-wrapper

Nimaro76, 2023-07-19:
Thanks, I will try it later. for now , I tried building the librealsense using cmake GUI and the rs-kinfu instructions and these were the errors I got when I wanted to build : /usr/bin/ld: cannot find -lvtkRenderingOpenGL2
/usr/bin/ld: cannot find -lvtkCommonCore
/usr/bin/ld: cannot find -lvtksys
/usr/bin/ld: cannot find -lvtkCommonDataModel
/usr/bin/ld: cannot find -lvtkCommonMath
/usr/bin/ld: cannot find -lvtkCommonMisc
/usr/bin/ld: cannot find -lvtkCommonSystem
/usr/bin/ld: cannot find -lvtkCommonTransforms
/usr/bin/ld: cannot find -lvtkCommonExecutionModel
/usr/bin/ld: cannot find -lvtkIOImage
/usr/bin/ld: cannot find -lvtkDICOMParser
/usr/bin/ld: cannot find -lvtkmetaio
/usr/bin/ld: cannot find -lvtkImagingCore
/usr/bin/ld: cannot find -lvtkRenderingCore
/usr/bin/ld: cannot find -lvtkCommonColor
/usr/bin/ld: cannot find -lvtkCommonComputationalGeometry
/usr/bin/ld: cannot find -lvtkFiltersCore
/usr/bin/ld: cannot find -lvtkFiltersGeneral
/usr/bin/ld: cannot find -lvtkFiltersGeometry
/usr/bin/ld: cannot find -lvtkFiltersSources
/usr/bin/ld: cannot find -lvtkInteractionStyle
/usr/bin/ld: cannot find -lvtkFiltersExtraction
/usr/bin/ld: cannot find -lvtkFiltersStatistics
/usr/bin/ld: cannot find -lvtkImagingFourier
/usr/bin/ld: cannot find -lvtkalglib
/usr/bin/ld: cannot find -lvtkRenderingLOD
/usr/bin/ld: cannot find -lvtkFiltersModeling
/usr/bin/ld: cannot find -lvtkIOPLY
/usr/bin/ld: cannot find -lvtkIOCore
/usr/bin/ld: cannot find -lvtkFiltersTexture
/usr/bin/ld: cannot find -lvtkRenderingFreeType
/usr/bin/ld: cannot find -lvtkIOExport
/usr/bin/ld: cannot find -lvtkRenderingGL2PSOpenGL2
/usr/bin/ld: cannot find -lvtkIOGeometry
/usr/bin/ld: cannot find -lvtkIOLegacy
/usr/bin/ld: cannot find -lIconv::Iconv
collect2: error: ld returned 1 exit status
make[2]: *** [wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/build.make:217: wrappers/opencv/imshow/rs-imshow] Error 1
make[1]: *** [CMakeFiles/Makefile2:1169: wrappers/opencv/imshow/CMakeFiles/rs-imshow.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

And I don't understand it. I checked that I installed vtk and I checked having eigen 3 too.  I would appreciate your help.

MartyG-RealSense, 2023-07-20:
As far as I can tell, vtk is not a requirement of the modern librealsense2 SDK and these errors have not been previously reported by users who were installing rs-kinfu.

Nimaro76, 2023-07-26:
issue resolved , thank you.

closed 2023-07-26 by Nimaro76

MartyG-RealSense, 2023-07-26:
You are very welcome.  I'm pleased to hear that you achieved a resolution.  Thanks very much for the update!