realsenseai/librealsense #12017: Retrieve depth data from facial landmark using pyrealsense2

HasarinduPerera, 2023-07-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 22  | 
| Kernel Version (Linux Only)    |  5.19.0-46-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python3                          | 
| Segment			|                            | 

### Issue Description

Hi there, I am just wondering how can I get the depth data to create an anti-spoofing face detector. I am currently using `dlib` to detect the face and determine the facial landmarks. What I want is to use the depth of the center of the face (around nose) and the depth of the edge of the face (around ear) and check whether center depth is lower than the edge. I have found the below wrapper but there is no python version for it. Any idea how to implement that using depth data?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/dlib/face

This is my current code:
```
# import the necessary packages
from imutils import face_utils
import dlib
import cv2
 
# initialize dlib's face detector (HOG-based) and then create
# the facial landmark predictor
p = "shape_predictor_68_face_landmarks.dat"
detector = dlib.get_frontal_face_detector()
predictor = dlib.shape_predictor(p)

cap = cv2.VideoCapture(0)
 
while True:
    # load the input image and convert it to grayscale
    _, image = cap.read()
    gray = cv2.cvtColor(image, cv2.COLOR_BGR2GRAY)
        
    # detect faces in the grayscale image
    rects = detector(gray, 0)
    
    # loop over the face detections
    for (i, rect) in enumerate(rects):
        # determine the facial landmarks for the face region, then
        # convert the facial landmark (x, y)-coordinates to a NumPy
        # array
        shape = predictor(gray, rect)
        shape = face_utils.shape_to_np(shape)
    
        # loop over the (x, y)-coordinates for the facial landmarks
        # and draw them on the image
        for (x, y) in shape:
            cv2.circle(image, (x, y), 2, (0, 255, 0), -1)
    
    # show the output image with the face detections + facial landmarks
    cv2.imshow("Output", image)
    k = cv2.waitKey(5) & 0xFF
    if k == 27:
        break

cv2.destroyAllWindows()
cap.release()

```
TIA

MartyG-RealSense, 2023-07-19:
Hi @HasarinduPerera  There is not an equivalent official face landmark detection example for Python, though a RealSense team member in a past question about this suggested googling for the search term below (including the dashes between words):  

**facial-landmarks-dlib-opencv-python**

HasarinduPerera, 2023-07-19:
Thank you for your response @MartyG-RealSense. My issue is that I am trying to access the depth of given two points using x,y coordinates, is there any method to retrieve that data using `pyrealsense2`?

HasarinduPerera, 2023-07-19:
Found a solution, Thank you!

```
# Get depth value at a specific x-y coordinate
x = 320  
y = 240 
depth_value = depth_frame.get_distance(x, y)
```

closed 2023-07-19 by HasarinduPerera