realsenseai/librealsense #12020: Align already captured RGB and depth images

FourierMourier, 2023-07-19:
### Issue Description
Hi, guys. I've searched through issues but haven't found any working example for the following: I already have captured color and depth images (as .jpg) and i need to align them. I have intrinsics and extrinsics data (and some additional depth info) so that I can initialize classes but I don't know how to properly use them - all I've seen so far is processing images from the stream, not from the file.
Can please someone navigate me or link the similar problem?

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/NVIDIA Jetson/ etc..  |
| SDK Version                            |  2.30+                        | 
| Language                            | python                         | 
| Segment			|  others                        |

MartyG-RealSense, 2023-07-19:
Hi @FourierMourier  The best available reference for alignment of already created image files is at https://github.com/IntelRealSense/librealsense/issues/1274

If you still have the camera then it would be easier to record depth and color streams into a **bag** file (which is like a video recording of camera data) and then load the bag file into the RealSense SDK and apply alignment to the stored data in real-time.

MartyG-RealSense, 2023-07-27:
Hi @FourierMourier  Do you require further assistance with this case, please?  Thanks!

FourierMourier, 2023-07-27:
@MartyG-RealSense Hi, thanks. I would like to, but not sure whether smth would help...btw, do you know why i pass rs.software_video_frame object to depth_sensor being of type rs.software_sensor in method .on_video_frame and get "Process finished with exit code -1073741819 (0xC0000005)" in the debug/run mode?

this is how i create such frame
`
    depth_video_frame = rs.software_video_frame()
    depth_video_frame.pixels = depth_frame
    depth_video_frame.stride = stride
    depth_video_frame.bpp = bpp
    depth_video_frame.timestamp = 0.0  # 0.0 as default
    depth_video_frame.domain = rs.timestamp_domain.hardware_clock
    depth_video_frame.frame_number = frame_number 
`
but segmentation fault happens anyway; also, i noticed that `pixels` attribute is expected of type rs.BufData but i can't use the constructor; `depth_frame` is a contiguous flatten np array of type np.uint16
what's the recommended way of creating rs.software_video_frame objects? how can i fill data then?

MartyG-RealSense, 2023-07-28:
Hi @FourierMourier  A RealSense team member provides advice about BufData at https://github.com/IntelRealSense/librealsense/issues/8394

My knowledge of software-device programming is admittedly limited and so I do not have advice that I can provide about **rs.software_video_frame()** unfortunately, though https://github.com/IntelRealSense/librealsense/issues/7057 has an example of a Python software-device script that uses it.

```
for i in [f for f in os.listdir() if 'npy' in f]:
    print('loading frame i')
    depth_npy = np.load(i, mmap_mode='r')

    vid_frame = rs.software_video_frame()
    vid_frame.stride = depth_stream.width * depth_stream.bpp
    vid_frame.bpp = depth_stream.bpp
    vid_frame.timestamp = 0.0
    vid_frame.pixels = depth_npy
    vid_frame.domain = rs.timestamp_domain.hardware_clock
    vid_frame.frame_number = int(i.split('.')[0])
    vid_frame.profile = dstream_prof
    vid_frame.pixels = depth_npy
    print('created software frame')
```

FourierMourier, 2023-07-28:
@MartyG-RealSense i tried both numpy array assignment for .pixels attribute as well as sharing data but
1. example listed in #8394 issue doesn't work with the same error (constructor is not defined)
2. i'm still getting seg fault errors when trying to put frames 
```
    depth_sensor.on_video_frame(depth_video_frame)
    color_sensor.on_video_frame(color_video_frame)
```
3. using the way in #7057 neither helps - i got the same error as the one back in 2020 (no device connected) using newest pylib version...

i already use software_device and syncer classes - cpp code works ok but python breaks due to internal errors; seems strange that 2 years later there's still no python constructor for rs.BufData object...

May i ask you whether the team member can provide a minimal example of how to properly use/initialize such objects?

MartyG-RealSense, 2023-07-29:
I will highlight your question to my Intel RealSense colleagues.  Thanks very much for your patience.

FourierMourier, 2023-07-30:
@MartyG-RealSense Thanks in advance

gwen2018, 2023-08-11:
@FourierMourier Thanks for raising the issue to the realsense team. Attached please find a sample python script to demonstrate construct software device with pre-captured images and align depth to color. This example first captures depth and color images from realsense camera into npy files and then create software device and align depth to color with the saved images. It was tested on Windows 10 with Python 3.9.10 and pyrealsense2-2.54.1.5217.

[align-with-software-device-example.py.txt](https://github.com/IntelRealSense/librealsense/files/12324155/align-with-software-device-example.py.txt)

FourierMourier, 2023-08-11:
@gwen2018 Thank you very much! I adjusted my data to your given blocks of the code and it finally works (even though i haven't managed to test full pipeline with project_points yet but I can see aligned images without any seg faults in py code)
environment:
win11/win10
pyrealsense2==2.53.1.4623
python 3.9.16
opencv-python==4.8.0.76

Some possible improvements as the comments in the code:
if `depth_npy` is not loaded as numpy array but as image and then is processed, it's expected to be a contiguous flatten array (from width*height) - almost the same applies to `color_npy` (numpy's .flatten() method seems to handle that properly without manual for loops)

I think this example 100% should be added to lib examples :)

MartyG-RealSense, 2023-08-12:
@FourierMourier  It's great to hear that the code kindly provided by @gwen2018 worked for you.

@gwen2018  Thanks so much for the help provided!

songdedaoyi, 2023-09-13:
Hi, I had the same problem with you and I am wondering whether you are willing to share your code to help me solve this problem?  I would appreciate your help so much!

FourierMourier, 2023-09-13:
@MartyG-RealSense Sorry for the late response. No, thank you and the team for the assistance! I managed to align to both (depth -> color and vice versa).
Just a small notice - alignment worked fine using 2.53 version (sometimes pypi cannot find version properly specified full version), then i decided to update and got nearly white images no matter what, and maybe 2 weeks after again normal images with the last version; so I'm not sure whether it's a particular package version issue or maybe the machine i worked on

MartyG-RealSense, 2023-09-13:
You are very welcome.  Thanks very much for the update about your success!

FourierMourier, 2023-09-13:
@songdedaoyi Hi! see the example here:
https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260

To proceed, please ensure that you properly initialize the intrinsics and extrinsics objects, following the format specified (since I don't know how you store the data). Additionally, if your depth image is in uint8 format (as image rather than smth like np.array), you will need to convert it to uint16. Once these preparations are complete, you can use the provided code from the Realsense team (link to the answer above).

FourierMourier, 2023-09-13:
@MartyG-RealSense By the way, this question is unrelated to the previous issue, but I'm curious: do depth cameras capture objects faster than color cameras? I've noticed that for fast-moving objects, the depth image appears clearer while the color image appears blurred.

MartyG-RealSense, 2023-09-14:
Hi @FourierMourier  RealSense camera models of the D435 type, and also D415, have a slow 'rolling' shutter on the RGB sensor and a fast 'global' shutter for the depth sensors.  So on those models with a rolling shutter, RGB blurring can occur during fast motion.  

The D455 type models have a fast global shutter on both the RGB and depth sensors.

Increasing RGB FPS to 60 whilst having depth FPS at 30 can help to reduce blurring on D415 / D435x models.

An alternative approach to blur reduction is to set RGB auto-exposure to manual and use a manual exposure value of '78' and an FPS of '6' (not 60).

songdedaoyi, 2023-09-14:
> @songdedaoyi Hi! see the example here: [#12020 (comment)](https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260)
> 
> To proceed, please ensure that you properly initialize the intrinsics and extrinsics objects, following the format specified (since I don't know how you store the data). Additionally, if your depth image is in uint8 format (as image rather than smth like np.array), you will need to convert it to uint16. Once these preparations are complete, you can use the provided code from the Realsense team (link to the answer above).

Thanks for your reply! My depth images are in uint16 format and saved as .tif. I am still confused because the sample python script they provide seems to need to capture frames first, I actually have no idea how to deal with the frames I had already captured to replace the function ’wait_for_frames()‘. Could you tell me how to deal with that?

FourierMourier, 2023-09-14:
@MartyG-RealSense Thanks for the info! Yeah, I also noticed on some d415 cameras this issue like depth was able to capture the objects and rgb wasn't - just wondering whether the fps reduction will lead to blurred objects on depth also... Thanks for the exposure adjustment feature - should've tried to check that : )

MartyG-RealSense, 2023-09-14:
You are very welcome!  Depth should not be affected by RGB blurring due to the fast shutter on the depth sensor.  When depth and RGB are aligned though, the position of the RGB color data may noticably lag behind the position of the depth data instead of both of them moving perfectly together if measures to reduce RGB lag are not taken.

FourierMourier, 2023-09-29:
Hi @MartyG-RealSense No, you guys helped me a lot so i believe no big issues left. Thanks for your assistance! It seems I have to close it : )

closed 2023-09-29 by FourierMourier

MartyG-RealSense, 2023-09-30:
You are very welcome.  I'm pleased that we could help.  Thanks very much for the update!

akashajagekar55, 2023-10-25:
Hello, I had the same issue that @FourierMourier  had. Basically, I have an RGB-D pair that is saved on my drive in png format. The depth image is Uint16, and the RGB image is Uint8 (I believe these are the correct formats). I followed the steps mentioned in this thread, but I am getting the following error:

Traceback (most recent call last):
  File "/home/akadsh/Desktop/demo.py", line 111, in <module>
    depth_frame = rs.software_video_frame(depth_data)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pyrealsense2.pyrealsense2.software_video_frame()

Invoked with: array([[   0,    0,    0, ...,    0,    0,    0],
       [   0,    0,    0, ...,    0,    0,    0],
       [   0,    0,    0, ..., 1809, 1804,    0],
       ...,
       [   0,    0,    0, ..., 2460, 2460,    0],
       [   0,    0,    0, ..., 2460, 2469,    0],
       [   0,    0,    0, ...,    0,    0,    0]], dtype=uint16)
       
       
I am not able to figure out where I am going wrong. All of my code is running, but the depth_frame = rs.software_video_frame(depth_data) line is giving me an error. Below, I have attached my full code:

`import cv2
import pyrealsense2 as rs
import numpy as np

fps = 30 

camera_depth_intrinsics          = rs.intrinsics()  # camera depth intrinsics
camera_color_intrinsics          = rs.intrinsics()  # camera color intrinsics
camera_depth_to_color_extrinsics = rs.extrinsics()  # camera depth to color extrinsics

sdev = rs.software_device()

depth_sensor: rs.software_sensor = sdev.add_sensor("Depth")

depth_intrinsics = rs.intrinsics()

depth_intrinsics.width  = camera_depth_intrinsics.width
depth_intrinsics.height = camera_depth_intrinsics.height

depth_intrinsics.ppx = camera_depth_intrinsics.ppx
depth_intrinsics.ppy = camera_depth_intrinsics.ppy

depth_intrinsics.fx = camera_depth_intrinsics.fx
depth_intrinsics.fy = camera_depth_intrinsics.fy

depth_intrinsics.coeffs = camera_depth_intrinsics.coeffs       ## [0.0, 0.0, 0.0, 0.0, 0.0]
depth_intrinsics.model = camera_depth_intrinsics.model

depth_stream = rs.video_stream()
depth_stream.type = rs.stream.depth
depth_stream.width = depth_intrinsics.width
depth_stream.height = depth_intrinsics.height
depth_stream.fps = fps
depth_stream.bpp = 2                              # depth z16 2 bytes per pixel
depth_stream.fmt = rs.format.z16
depth_stream.intrinsics = depth_intrinsics
depth_stream.index = 0
depth_stream.uid = 1

depth_profile = depth_sensor.add_video_stream(depth_stream)

color_sensor: rs.software_sensor = sdev.add_sensor("Color")

color_intrinsics = rs.intrinsics()
color_intrinsics.width = camera_color_intrinsics.width
color_intrinsics.height = camera_color_intrinsics.height

color_intrinsics.ppx = camera_color_intrinsics.ppx
color_intrinsics.ppy = camera_color_intrinsics.ppy

color_intrinsics.fx = camera_color_intrinsics.fx
color_intrinsics.fy = camera_color_intrinsics.fy

color_intrinsics.coeffs = camera_color_intrinsics.coeffs
color_intrinsics.model = camera_color_intrinsics.model

color_stream = rs.video_stream()
color_stream.type = rs.stream.color
color_stream.width = color_intrinsics.width
color_stream.height = color_intrinsics.height
color_stream.fps = fps
color_stream.bpp = 3                                # color stream rgb8 3 bytes per pixel in this example
color_stream.fmt = rs.format.rgb8
color_stream.intrinsics = color_intrinsics
color_stream.index = 0
color_stream.uid = 2

color_profile = color_sensor.add_video_stream(color_stream)

depth_to_color_extrinsics = rs.extrinsics()
depth_to_color_extrinsics.rotation = camera_depth_to_color_extrinsics.rotation
depth_to_color_extrinsics.translation = camera_depth_to_color_extrinsics.translation
depth_profile.register_extrinsics_to(depth_profile, depth_to_color_extrinsics)

depth_sensor.open(depth_profile)
color_sensor.open(color_profile)

camera_syncer = rs.syncer()
depth_sensor.start(camera_syncer)
color_sensor.start(camera_syncer)

align_to = rs.stream.color
align = rs.align(align_to)

colorizer = rs.colorizer()

depth_img = cv2.imread('/home/akadsh/Downloads/depthimg_2023-04-25 15_21_05.png', cv2.IMREAD_UNCHANGED)
depth_data = np.array(depth_img)
rgb_img = cv2.imread('/home/akadsh/Downloads/image_2023-04-25 15_21_05.png')
rgb_data = np.array(rgb_img) 

depth_frame = rs.software_video_frame(depth_data)
rgb_frame = rs.software_video_frame(rgb_data)

aligned_frames = align.process(depth_frame, rgb_frame)`
Any help would be appreciated.

matteomastrogiuseppe, 2025-08-05:
The code provided in https://github.com/IntelRealSense/librealsense/issues/12020#issuecomment-1675211260 works well, it really gets the job done, many thanks for that.

For reference, I had to add an extra line to make it run:
`depth_sensor.add_read_only_option(rs.option.depth_units, 0.001)` 

right before the `depth_sensor.open(depth_profile)`

(Tested on Ubuntu 24.04 with Python 3.11 and pyrealsense2-2.55.1.6486)