realsenseai/librealsense #12022: librealsense inside a docker still needs udev rules installed in the host

martinakos, 2023-07-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |  5.15.0.2 | 
| Operating System & Version |  Ubuntu 20.04/22.04  | 
| Kernel Version (Linux Only)    | 5.15.0-72-generic                                        | 
| Platform                                 | PC|
| SDK Version                            |  v2.54.1    | 
| RealSense ROS                     | v4.54.1 |

I built the librealsense SDK from source using the RSUSB backend. Set FORCE_RSUSB_BACKEND=True. 
I did this inside a docker container running Ubuntu 22.04 
Then I build Realsense ROS (ROS humble) using this RSUSB backend librealsense.  
I can run the realsense2_camera inside the docker fine and I get all the metadata from the image streams.

Then I go and I remove all realsense drivers from the **host** with:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

I reboot the machine and try to run again the realsense2_camera driver inside my docker. 
But now it doesn't work!    (it seems it needs something in the host to work inside a docker?)

In the host I copy librealsense/config/99-realsense-libusb.rules to /etc/udev/rules.d/
Then I reload the udev rules in the host with 

`udevadm control --reload-rules && udevadm trigger
`

And try again the  realsense2_camera driver inside my docker. 
Now it works again!!!

Is there anyway of running librealsense inside a docker avoiding having to copy this 99-realsense-libusb.rules file in the host?

I  thought that building librealsense with the RSUSB backend would allow me to install it in a docker container without having to install anything in the host (In contrast with using the V4L backend which will require the librealsense2-dkms installed in the host so that it can work inside the docker).  However, now I see that I have to copy a 99-realsense-libusb.rules file in the host /etc/udev/rules.d for librealsense to work  inside a docker.

MartyG-RealSense, 2023-07-20:
Hi @martinakos  In the official Docker installation tutorial for librealsense provided by Intel at the link below, their Dockerfile also copies /etc/udev/rules.d/ so it does seem to be a requirement.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker
https://github.com/IntelRealSense/librealsense/blob/master/scripts/Docker/Dockerfile#L61

martinakos, 2023-07-21:
That is copying the 99-realsense-libusb.rules file inside the docker image. 
My surprise is that it seems that this file needs to be copied in the host machine too, which breaks the deployment scenario I wanted to solve by using docker. Any way around it?

MartyG-RealSense, 2023-07-21:
It is frequent practice by RealSense users who use Docker to have two librealsense installations in a Docker setup, one on the host and another in the Docker container.  The host and Docker installations are allowed to have different versions of the librealsense SDK, ROS wrapper and Ubuntu though.

My recollection however is that this tends to be because users have a project where they need to use a particular setup in a container for a certain project but do not want to use that setup on the main host machine (for example, having the RealSense ROS1 wrapper on the Docker but using the ROS2 wrapper on the host).  So the impression that I get is that having librealsense on the host machine is optional (my Docker knowledge is admittedly limited).

The udev rules are installed by the librealsense SDK and not the ROS wrapper.  If librealsense is built from packages then it installs the udev rule to **/lib/udev/rules.d/60-librealsense2-udev-rules.rule**.  However, if librealsense is installed from source code then it installs a different udev rule, at **/etc/udev/rules.d/99-realsense-libusb.rules**.

If both a package and source code installation of librealsense is performed on the same computer then you can end up having two conflicting udev rules installed, the 60 and 99 ones.  In that situation, the rule that should be deleted is **/etc/udev/rules.d/99-realsense-libusb.rules** because deleting the /lib version of the rule breaks linkages to the debian packages.

closed 2023-08-02 by MartyG-RealSense

MartyG-RealSense, 2025-05-02:
Thanks so much @ivoshopov-uv for sharing this information!