realsenseai/librealsense #12024: L515 Unable to obtain a valid depth value by depth image

le-wei, 2023-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (v2.53.1) | 
| Operating System & Version |   { Linux (Ubuntu 20.4)}   | 
| Kernel Version (Linux Only)    |  (5.15.0-76)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.53.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
Use the following code to get a depth image:
```
 config.enable_stream(RS2_STREAM_DEPTH, 0, 1024, 768, RS2_FORMAT_Z16, 30);
....
    cv::Mat depth_image(cv::Size(depth_w, depth_h), CV_16U,
                        (void *)aligned_depth_frame.get_data(),
                        cv::Mat::AUTO_STEP);
```

Then use the following code to get the depth value of the depth image:

`float temdisp = depthImage.at<uint16_t>(y, x) * (1.0 / depthFactor);`
The value obtained by L515 is wrong, D435i is able to obtain the correct value.
Please help. Thank you very much.

MartyG-RealSense, 2023-07-21:
Hi @le-wei  If you are multiplying uint16_t by the depth unit scale and using the same value for both cameras then the L515 result will be incorrect.  This is because the two camera models use different scale values.  D435i's is **0.001** and L515's is **0.000250**

le-wei, 2023-07-21:
Thank you very much for your quick answer, it has solved my problem!

closed 2023-07-21 by le-wei

MartyG-RealSense, 2023-07-21:
You are very welcome.  I'm pleased that I could help.  :)