realsenseai/librealsense #12026: Displaying aligned color & depth image in matlab

chhayank-sri, 2023-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L500 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Language                            | Matlab                         | 
| Segment			|  Robot                      | 

### Issue Description
I am using matlab to process data obtained from L500 camera.
The idea is to extract image data from *.bag file and display it with (imshow).

I am using the below code which has been derived from the available examples (rosbag_example.m,  & depth_example.m), Now I dont want to use the colorizer to color the depth frame instead align depth to color (Depth -> color) frame.

Below is the code
```matlab
% Make Config object to manage pipeline settings
cfg = realsense.config();
validateattributes(filename, {'char','string'}, {'scalartext', 'nonempty'}, '', 'filename', 1);
% Tell pipeline to stream from the given rosbag file
cfg.enable_device_from_file(filename)

% Make Pipeline object to manage streaming
pipe = realsense.pipeline();

% Start streaming from the rosbag with default settings
profile = pipe.start(cfg);

% Get streaming device's name
dev = profile.get_device();
name = dev.get_info(realsense.camera_info.name);

% Align to color frame
align = realsense.align(realsense.stream.color());

fs = pipe.wait_for_frames();
fs = align.process(fs);

color_frame = fs.get_color_frame();
color_data = color_frame.get_data();

depth_frame = fs.get_depth_frame();
depth_data   = depth_frame.get_data();

pipe.stop();
``` 
This generates variables as shown 
[image]

Now the question is - : 
color_frame.get_width() => 640
color_frame.get_height() => 480
But color_data = color_frame.get_data() is of dimension 1x153600 uint32. (should be 1x307200x3) for 480x640x3 color image

1) In which step there seems to be a mistake to get color_frame's data ? Is any other way color_data should be processed for viewing in matlab? 
2) Also imshow requires data in uint8,uint16,double but not uint32 could I directly convert the data to double
3) on comparison with python, the array values also differs between python script and matlab (I have checked for the same frame number using get_frame_number() method), why is that?

Equivalent code in python uses np.asanyarray(color_frame.get_data()) generates as (480,640) arrray 
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

MartyG-RealSense, 2023-07-21:
Hi @chhayank-sri  Whilst I do not know the answers t your three questions, there is a MATLAB depth-color alignment script at https://github.com/IntelRealSense/librealsense/issues/4245#issuecomment-503552164 which may be a helpful reference if you have not seen it already.

chhayank-sri, 2023-07-21:
Hi @MartyG-RealSense , I did see the comment previously, he has used a similar approach only he is not using a pre-recorded bag file (while it should not cause any issue) but it has the same size issue. 
[image]

chhayank-sri, 2023-07-21:
I have another *.bag file recorded from D455 file, I am able to use the data from D455 but there seems to be an issue with L515. Idk if that helps!!

MartyG-RealSense, 2023-07-22:
Which firmware driver version is your L515 camera using, please?  It should be updated to the latest available one, **1.5.8.1**.  L515 did not have support for 640x480 color in its early firmware versions and the feature was added later.

MartyG-RealSense, 2023-07-29:
Hi @chhayank-sri  Do you require further assistance with this case, please?  Thanks!

chhayank-sri, 2023-07-30:
Hi @MartyG-RealSense , I am using the latest firmware. The data recording and processing was done with the same version.

For Image with YUYV encoding
color_frame.get_width() => 640
color_frame.get_height() => 480
I figured out why the color_data was of the dimension 1x153600 uint32. (These are macro-pixel values)

Since the encoding is YUYV - for each 32-bit macro-pixel corresponding Luma (Y) and Chrome (U/V) of 8-bits need to be extracted.

(Macro-pixel (32bit) ==> Y0 (8bit) | U0 (8bit) | Y1 (8bit) | V0 (8bit) )
Source: https://www.cs.auckland.ac.nz/courses/compsci773s1c/lectures/YuY2.htm

I have added by solution if anyone else is interested in using the Matlab wrapper and is facing a similar issue.
https://github.com/chhayank-sri/Bag_data_extraction.git

Thank you for your help.

closed 2023-07-30 by chhayank-sri

MartyG-RealSense, 2023-07-30:
Thanks so much @chhayank-sri for the excellent news that you achieved a solution and for sharing the solution with the RealSense community!