realsenseai/librealsense #12027: Error during building librealsense2 in RPI4 with Raspberry PI OS x64

Harishkumarnimmala, 2023-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
**Issue at which command ?**

cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true

**What is the error ?**

[ 56%] Building CXX object examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o
[ 56%] Linking CXX executable rs-hello-realsense
/usr/bin/ld: ../../librealsense2.so.2.54.1: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.54.1: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.54.1: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:93: examples/hello-realsense/rs-hello-realsense] Error 1
make[2]: Leaving directory '/home/pi/librealsense/build'
make[1]: *** [CMakeFiles/Makefile2:525: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make[1]: Leaving directory '/home/pi/librealsense/build'
make: *** [Makefile:130: all] Error 2

MartyG-RealSense, 2023-07-23:
Hi @Harishkumarnimmala  At https://github.com/IntelRealSense/librealsense/issues/10664#issuecomment-1183204808 there is information about the **undefined reference to __atomic_store_8** error and what a RealSense Pi user who experienced that user did to resolve it.

MartyG-RealSense, 2023-08-04:
Case closed due to no further comments received.

closed 2023-08-04 by MartyG-RealSense

AnoukDeBock, 2024-03-27:
Hello, I have encountered the same error. I am trying to build the SDK library with Raspbian on Raspberry Pi 3+

/usr/bin/ld: ../../Release/librealsense2.so.2.54.2: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../Release/librealsense2.so.2.54.2: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../Release/librealsense2.so.2.54.2: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:109: Release/rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/Makefile2:1054: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

I already looked up all issues regarding this topic but nothing seems to work. 
Things I have tried: 
- set(CMAKE_CXX_LINK_FLAGS "${CMAKE_CXX_LINK_FLAGS} -latomic")
- installing gcc 
- sudo apt install clang llvm

Does anybody have a suggestion to fix this error? 

Thanks you in advance!

MartyG-RealSense, 2024-03-28:
Hi @AnoukDeBock  Have you tried the Raspbian installation instructions for Pi 3B at the link below, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

If you have the option of using Ubuntu instead of Raspbian then the **libuvc** method at the link below typically works well with Raspberry Pi.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md