realsenseai/librealsense #12028: camera realsense D435 &ublox  time synchronization

Mahmoudadham4040, 2023-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | realsense D435 | 
| Operating System & Version |   Ubuntu 18  | 
| Platform                                 | PC/Raspberry  |

### Issue Description
Could you please inform me if the real sense D435 camera can be synchronized (hardware-based camera pins) with GPS receivers (i.e. Ublox)? 
(I asked about other devices such as GPS, lidar, and imu, not multi-cameras synchronization).

MartyG-RealSense, 2023-07-24:
Hi @Mahmoudadham4040  You can configure the D435 camera to output a trigger pulse from pin 5 (the Sync pin) of its GPIO sync pin set, which can be found under a small removable cap on the top of the camera casing.  

https://github.com/IntelRealSense/librealsense/issues/8171 has an example of C++ code for setting the RealSense SDK option **RS2_OPTION_OUTPUT_TRIGGER_ENABLED** to '1' to enable the trigger pulse.

   ```
 if (depth_sensor.supports(RS2_OPTION_OUTPUT_TRIGGER_ENABLED))
    {
        std::cout<<"here";
        depth_sensor.set_option(RS2_OPTION_OUTPUT_TRIGGER_ENABLED, 1.f); // Enable trigger
    }
```

The trigger pulse can also be activated in the RealSense Viewer tool by enabling the **Output Trigger Enabled** option which can be found in the Viewer's options side-panel under the 'Stereo Module > Controls' section.

[image]

You could conceivably run a wire from Pin 5 on the camera's sync pin set to a pin on the non-camera device so that the trigger pulse that is output from the camera travels outwards from the camera to your external device.

If your external device can output a trigger signal then it is also possible to configure it the reverse way - to have the external device act as the 'master' and transmit a trigger through a wire to Pin 5 on the camera when it is configured as a 'slave' that will attempt to follow the timing of the external device's trigger pulse.

Mahmoudadham4040, 2023-07-26:
Thanks for your response, i did all you stated but i want to check if the synchronization between the ublox and realsense is success . Is there any way to check?

MartyG-RealSense, 2023-07-26:
The RealSense camera should be configured for hardware sync with the option **Inter Cam Sync Mode**.  This option should be set to '1' if your RealSense camera is acting as master (the generator of the trigger pulse) or set to '2' if it is a slave (receiving a trigger pulse from your ublox).

If your ublox produces a timestamp then comparing that timestamp with the RealSense camera's timestamp could show whether sync is occurring.  If the two timestamps remain identical over time then hardware sync is **not** successful.  If the timestamps very slowly drift apart over tens of minutes then sync **is** successful.  This phenemon, which is the opposite of what would be expected, is described at the link below in the section headed **Now to the somewhat counter intuitive aspect of time stamps**.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#3-multi-camera-programming

Another method of validating sync is to use a **high speed LED panel** product.

https://support.intelrealsense.com/hc/en-us/community/posts/360049401673-Multi-sensor-synchronization-validation