realsenseai/librealsense #12030: Title: Manual Focus Control for Intel RealSense D435 Camera on Jetson Orin

ronbitonn, 2023-07-24:
I have a Jetson Orin development board and have connected an Intel RealSense D435 camera to it. My research task involves recognizing individual small ruminants by analyzing RGB and depth frames captured by the D435 camera. Specifically, I want to analyze the facial features of the small ruminants and perform text spotting to recognize the numbers on their ear tags.

The D435 camera is positioned approximately 1 meter away from the face of the small ruminant. However, I have encountered issues with the camera's focus, as not all frames look well due to some being out of focus.

I am utilizing the pyrealsense2 library for interacting with the D435 camera, which allows me to stream data from the camera and access its functionalities.

To ensure I can perform the analysis accurately, I am seeking assistance in implementing manual focus control for the D435 camera using pyrealsense2 on Jetson Orin. With manual focus control, I hope to adjust the camera's focus settings programmatically, ensuring that the captured frames are consistently in focus and suitable for my analysis.

Any guidance, code examples, or resources on how to do manual focus control for the D435 camera on Jetson Orin using the pyrealsense2 library would be greatly appreciated.

Thank you,
Ron

ronbitonn, 2023-07-24:
In the documentation, we can find these functions. maybe they can help me with my issue?
[](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.roi_sensor.html#pyrealsense2.roi_sensor.set_region_of_interest)

<p class="rubric" style="margin-top: 30px; font-weight: bold; text-align: justify; line-height: 20.8px; color: rgb(0, 0, 0); font-family: sans-serif; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Methods</p>

__init__((self: pyrealsense2.roi_sensor, ...) |  
-- | --
as_color_sensor(...) |  
as_depth_sensor(...) |  
as_fisheye_sensor(...) |  
as_motion_sensor(...) |  
as_pose_sensor(...) |  
as_roi_sensor(...) |  
as_wheel_odometer(...) |  
close((self: pyrealsense2.sensor) -> None) | Close sensor for exclusive access.
get_info((self: pyrealsense2.sensor, ...) | Retrieve camera specific information, like versions of various internal components.
get_option((self: pyrealsense2.options, ...) | Read option value from the device.
get_option_description(...) | Get option description.
get_option_range(...) | Retrieve the available range of values of a supported option
get_option_value_description(...) | Get option value description (In case a specific option value holds special meaning)
get_recommended_filters(...) | Return the recommended list of filters by the sensor.
get_region_of_interest(...) |  
get_stream_profiles(...) | Retrieves the list of stream profiles supported by the sensor.
get_supported_options(...) | Retrieve list of supported options
is_color_sensor(...) |  
is_depth_sensor(...) |  
is_fisheye_sensor(...) |  
is_motion_sensor(...) |  
is_option_read_only(...) | Check if particular option is read only.
is_pose_sensor(...) |  
is_roi_sensor(...) |  
is_wheel_odometer(...) |  
open(*args, **kwargs) | Overloaded function.
set_notifications_callback(...) | Register Notifications callback
set_option((self: pyrealsense2.options, ...) | Write new value to device option
set_region_of_interest(...) |  
start(*args, **kwargs) | Overloaded function.
stop((self: pyrealsense2.sensor) -> None) | Stop streaming.
supports(*args, **kwargs) | Overloaded function.

</body>
</html>Methods

[__init__](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.roi_sensor.html#pyrealsense2.roi_sensor.__init__)((self: pyrealsense2.roi_sensor, ...)	
as_color_sensor(...)	
as_depth_sensor(...)	
as_fisheye_sensor(...)	
as_motion_sensor(...)	
as_pose_sensor(...)	
as_roi_sensor(...)	
as_wheel_odometer(...)	
close((self: pyrealsense2.sensor) -> None)	Close sensor for exclusive access.
get_info((self: pyrealsense2.sensor, ...)	Retrieve camera specific information, like versions of various internal components.
get_option((self: pyrealsense2.options, ...)	Read option value from the device.
get_option_description(...)	Get option description.
get_option_range(...)	Retrieve the available range of values of a supported option
get_option_value_description(...)	Get option value description (In case a specific option value holds special meaning)
get_recommended_filters(...)	Return the recommended list of filters by the sensor.
[get_region_of_interest](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.roi_sensor.html#pyrealsense2.roi_sensor.get_region_of_interest)(...)	
get_stream_profiles(...)	Retrieves the list of stream profiles supported by the sensor.
get_supported_options(...)	Retrieve list of supported options
is_color_sensor(...)	
is_depth_sensor(...)	
is_fisheye_sensor(...)	
is_motion_sensor(...)	
is_option_read_only(...)	Check if particular option is read only.
is_pose_sensor(...)	
is_roi_sensor(...)	
is_wheel_odometer(...)	
open(*args, **kwargs)	Overloaded function.
set_notifications_callback(...)	Register Notifications callback
set_option((self: pyrealsense2.options, ...)	Write new value to device option
[set_region_of_interest](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.roi_sensor.html#pyrealsense2.roi_sensor.set_region_of_interest)(...)	
start(*args, **kwargs)	Overloaded function.
stop((self: pyrealsense2.sensor) -> None)	Stop streaming.
supports(*args, **kwargs)	Overloaded function.

MartyG-RealSense, 2023-07-24:
Hi @ronbitonn  RealSense cameras do not have an adjustable focus control as the focus of the lenses is fixed.

The easiest way to resolve your focus issue may be to maximize the RGB **Sharpness** option to '100', which can make a significant difference to blurred images.  https://github.com/IntelRealSense/librealsense/issues/11246 has an example of Python code for configuring this option.

In that script, they first set up definitions for the depth and RGB (color) sensors.

`depth_sensor, color_sensor, *_ = profile.get_device().query_sensors()`

Then they use the color sensor definition to access the RGB sensor and configure the RGB sharpness.

`color_sensor.set_option(rs.option.sharpness, 100)`

ronbitonn, 2023-07-24:
Dear @MartyG-RealSense,

Thank you for your prompt response and valuable insight regarding the focus control for the Intel RealSense D435 camera on Jetson Orin.

I understand that RealSense cameras do not have adjustable focus control due to their fixed-focus lenses. However, I greatly appreciate your suggestion to maximize the RGB Sharpness option to '100' as a possible solution to improve the sharpness and clarity of the captured frames. I will try implementing the provided Python code example to configure the RGB sharpness and assess its impact on image quality.

Should I encounter any further issues or require additional assistance, I will reach out to you or the community on the "librealsense" repository on GitHub. Your expertise and guidance are immensely valuable to my research task.

Best regards,

Ron

MartyG-RealSense, 2023-07-24:
You are very welcome.  Please do feel welcome to ask further questions on this librealsense GitHub support forum if you need to do so.  Good luck!

MartyG-RealSense, 2023-07-31:
Hi @ronbitonn  Do you currently require further assistance with this case, please?  Thanks!

ronbitonn, 2023-08-01:
Dear @MartyG-RealSense,

I implemented the suggested approach to maximize the RGB Sharpness option, but some images still lack the necessary clarity to recognize the small ruminants' ear tags. As I continue my research, I'd appreciate any additional recommendations or alternative techniques to improve image quality and enhance ear tag recognition.

Your support has been invaluable, and I'm determined to find a solution for accurate analysis.

Thank you for your assistance.

MartyG-RealSense, 2023-08-01:
If you are seeking to further improve the readablity of text on the RGB image, does maximizing **gain** help?

color_sensor.set_option(rs.option.sharpness, 128)

ronbitonn, 2023-08-11:
Hello,

Thank you for your assistance and suggestions. I wanted to let you know that I'm actively working on improving the image quality of the Intel RealSense D435 camera for my research. I've begun by adjusting the camera's position to see if that has an impact on the clarity of the captured frames.

I truly appreciate the guidance you've provided so far, and I will keep you updated on the results of the changes I'm making. If I encounter any further challenges or if there are additional aspects where I might need assistance, I won't hesitate to reach out for help.

Once again, thank you for your support.

MartyG-RealSense, 2023-08-11:
You are very welcome.  Thanks very much for the update and please do keep us updated with your progress and findings.  Good luck!

closed 2023-10-19 by MartyG-RealSense