realsenseai/librealsense #12035: D435f camera projector not working for a particular unit

echoGee, 2023-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |d435f| 

### Issue Description
The projector patterns are not visible in the center of the image. This happens in one of the camera. Could it be a hardware malfunction?
[image]

MartyG-RealSense, 2023-07-26:
Hi @echoGee  I was not able to replicate this effect when pointing a D435f at the floor, I will highlight your issue to my Intel RealSense colleagues.  Thanks very much for your patience.

[image]

echoGee, 2023-07-26:
One of our camera is fine and works as expected. Second one shows the faulty behavior

MartyG-RealSense, 2023-07-26:
My RealSense colleagues confirmed that it appears to be a hardware defect and ask that you RMA return the camera to the retailer that you purchased it from.

closed 2023-08-02 by echoGee