realsenseai/librealsense #12037: D405 flickering imaging screen when switching hdr mode on sdk

Frank-Wang123, 2023-07-26:
Issue Description

I access D405 by Intel Realsense Viewer v2.54.1, or Depth Quality Tool v2.54.1, when I click 'Hdr Enabled' box in the list of 'Controls' , the imaging screen is flickering, I would like to know if it is the reason that: 
1) d405 does not support this function, 
or 2) a SDK bug, 
or 3) there is a hardware or software failure in the camera? 
Thank you!

MartyG-RealSense, 2023-07-26:
Hi @Frank-Wang123  Is it the infrared image that is flickering in HDR mode?  If it is then this is normal when HDR is enabled.  The flickering can be stopped by configuring a post-processing filter called **Filter By Sequence ID**, as described at https://github.com/IntelRealSense/librealsense/issues/10505#issuecomment-1126879337

Frank-Wang123, 2023-07-26:
Hi MartyG, thank you for your reply. Unfortunately, it is not only the IR image window that flickers, but also the rgb window when opened. Please check the attached. This camera is new, so I'm trying to determine if the problem is at the hardware level.

https://github.com/IntelRealSense/librealsense/assets/18051425/7dbe3767-724b-4f6f-9168-e7143ce133f0

MartyG-RealSense, 2023-07-26:
I confirmed that RGB on the D405 flickers when HDR is enabled.  This is likely because the D405 model does not have a separate RGB sensor and instead obtains RGB from the depth sensor.  This means that changes to the depth sensor settings also affect RGB.

An example of this is exposure.  Because D405 does not have a separate RGB exposure setting, increasing or decreasing depth exposure increases or reduces the brightness of the RGB image.

The post-processing settings at https://github.com/IntelRealSense/librealsense/issues/10505#issuecomment-1126879337 do not stop RGB and infrared from flickering on the D405 model though.

Frank-Wang123, 2023-07-26:
Thank you very much for your detailed reply！

Filip5M, 2023-07-27:
I am having a somewhat related issue with flickering but only when a recorded bag file is played back in the viewer. I am using Realsense Viewer v2.54.1 on a mac (10.15.7) and D430 Camera w/o RGB. 

I am able to setup the HDR merge, exposures and IDs so the live depth feed looks great, and IR frames are filtered properly. Once recorded, when I try to playback the file i can see that all the frame info seems to be as it should but both the IR and depth videos constantly flicker. It seems as neither the IR frames are being filtered nor is the HDR merge working when played back.

I have tried toggling the bag files HDR merge, and Filter by Sequence ID options in the viewer with no change.

Is this expected behavior or am i missing something?

MartyG-RealSense, 2023-07-28:
Hi @Filip5M  I tested the HDR settings with live streaming and with bag file and confirmed that whilst the flickering stopped on the live stream when the settings were applied, the flicker appeared on the bag file even though the anti-flicker settings were included in the bag recording.

Filip5M, 2023-07-28:
@MartyG-RealSense, thank you for confirming. 

Is there a way to enable at least the HDR Merge for the depth data when replaying a bag file, or is this some functionality that may be added in a future version of the viewer?

MartyG-RealSense, 2023-07-29:
If a bag was recorded without HDR being enabled at the time then it cannot be enabled in the Viewer during bag playback.  

No Viewer option under the 'Controls' section of the side-panel can be changed during bag playback.  Only the depth visualization and post-processing options can be used.  So the ability to enable HDR Merge for a bag if it was not recorded into the bag would not be added, as all of the options in the Controls section are non-interactive.

HDR Enabled and HDR Merge can be controlled with program scripting though.  So it is conceivable that a bag playback program could be written that enables HDR and the HDR Merge once playback has begun.

https://support.intelrealsense.com/hc/en-us/community/posts/360053181133--solved-D455-HDR-merge-C-code-example-needed

I performed further tests with the D405 and the Viewer.  A workaround for the flickering color image in HDR mode was to have Color disabled and instead enable Depth and set Infrared to the **RGB8** format instead of Y8.  When HDR is enabled and the Sequence ID filter applied, you then have a depth stream and a non-flickering color stream.  You cannot use Y8 infrared though when color is coming from the Infrared stream.

Filip5M, 2023-07-31:
Hi @MartyG-RealSense just to clarify I am unable to get the bag file playback in the viewer to merge the depth data and not flicker even after setting everything up related to HDR merge and sequence ID filtering before I record the file. 

The issue I am experiencing is once all the HDR merge options are setup, the live stream looks fine and I hit record and it continues to look fine (both depth and IR). Then when I go to playback the file my depth data looks very different and is constantly flickering and has a lot of holes. I can confirm that the HDR merge post processing is already on in the bag file and I cant turn it off. Also, regardless of what ID I select to filter and whether of not the filter by ID option is enabled in the playback, the depth and IR frames continue to flicker.

MartyG-RealSense, 2023-08-01:
I was not able to replicate flickering on the bag after it has been recorded with D405 with HDR enabled, the sequence set to 1, 2, 1 and the Filter By Sequence ID post-processing filter enabled.

When creating a bag, I recommend enabling HDR Enabled first and **then** setting the 1. 2. 1 sequence below it.  This is because if you set the sequence first and enale HDR Enabled secondly afterwards, Sequence ID (the last of the three values) automatically changes from '1' to 'UVC'.

closed 2023-08-16 by MartyG-RealSense