realsenseai/librealsense #12039: No frames received!

mariadelmar2497, 2023-07-26:
| Camera Model                       | R D435i  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Linux (Ubuntu 20.04.6 LTS)  | 
| Kernel Version (Linux Only)    |  5.4.0-1088-raspi                                       | 
| Platform                                 | Raspberry Pi4  |
| SDK Version                            | 2.54.1                       | 
| Language                            |  python                         | 
| Segment			|  Robot                         | 

### Issue Description
I get the following error 'No frames received!' when I activate the RGB Camera module in Realsense-viewer.

[image]

I get the following log:
synthetic-stream.cpp:51 - Exception was thrown during callback: rscuda9alloc_devI4int2EESt10shared_ptrIT_EiEUlPS1_E_

But if I try to use the Cheese app which is the default camera application in Ubuntu it works fine, it shows an image. 
If I try to use the pyrealsense2 lib in python it doesn't work because no frames are received. So I don't know what to do...

Please any help would be very appreciated.

MartyG-RealSense, 2023-07-26:
Hi @mariadelmar2497  What method did you use to install the librealsense SDK on your Pi 4, please?

mariadelmar2497, 2023-07-26:
Hi, I followed these instructions to install the librealsense SDK: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

MartyG-RealSense, 2023-07-26:
The librealsense SDK can be difficult to install on Raspberry Pi.  There is also a known issue where the IMU (accel and gyro) of camera models such as D435i and D455 may not be accessible.

Whilst building from source code with CMake, like in the guide that you used, is the appropriate method of installing the SDK on Pi, it will have the best chance of working on Pi if the flag **-DFORCE_RSUSB_BACKEND=TRUE** is included in the CMake build instruction.  This builds the SDK in RSUSB Backend mode, which is not dependent on Linux versions or kernel versions and does not require a patch to be applied to the kernel.

mariadelmar2497, 2023-07-28:
Hi, sorry for my late reply. I tried your suggestion, and everything stayed the same. But after that, I downgraded the Ubuntu version installed on my Raspberry Pi 4 from Ubuntu 20.04 to Ubuntu 18.04 and followed the same steps on the installation guide, adding your tag suggestion on the CMake command, and now it is working! So maybe switching to Ubuntu 18.04 was the solution.

Thank you so much for your help!

MartyG-RealSense, 2023-07-28:
You are very welcome.  It's excellent news that you achieved a solution.  Thanks very much for the update!

closed 2023-07-28 by mariadelmar2497