realsenseai/librealsense #12043: advance_mode load_json not working

kevinbriandIno, 2023-07-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                      | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  4.6.1.0                       | 
| Language                            |  {C++ }                          | 
| Segment			|  others                          | 

### Issue Description
Hey so I'm trying to use the load_json function from the advanced_mode but it doesn't seem to work as I don't see any changes on the images and when I call serialize_json my modifications from load_json aren't there.
So here is a step by step of what I'm doing:

1. I validate that the advanced_mode is enabled before calling the load_json method with `advanced_mode_dev.is_enabled()`. 
2. I validate that the json file exists and is in a valid format (using nlohmanJson library)
3. I call the load_json method using this (also tried loading the config with nlohmanJson library and calling the dump method):
`
std::string contents((std::istreambuf_iterator<char>(inputFile)), std::istreambuf_iterator<char>());
advanced_mode_dev.load_json(contents);
`
    - I get no errors or warning from this last step
4. (I download the json file here and I don't get my modifications)
5. I apply basic configs to the sensors
6. I start my pipeline
7. I grab images (for 10 seconds)
     -  I try to validate if the captured images are different or if I can seethe laser on the physical camera with no luck (when I modify my config to have laserstate to on)
8. (I download the json file here and I don't get my modifications)

Here is what my json file looks like:
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "-0",
    "aux-param-colorcorrection11": "-0",
    "aux-param-colorcorrection12": "-0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "-0",
    "aux-param-colorcorrection6": "-0",
    "aux-param-colorcorrection7": "-0",
    "aux-param-colorcorrection8": "-0",
    "aux-param-colorcorrection9": "-0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "33000",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "1",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "off",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "480",
    "stream-width": "640"
}

I also tried using the file produced by serialize_json and only modifying parameters like laserstate to on with no success. 

Thank you. 
- Kevin

MartyG-RealSense, 2023-07-27:
Hi @kevinbriandIno  It looks as though most of the values in the json are the default values for those options, so if those values are the defaults anyway then applying the json would not make much difference.

If you are setting a custom resolution (640x480) then the json will not change this, because when the pipeline starts it will override the json's resolution settings.  A way around this is to load in the json and then import the json's **stream-width** and **stream-height** resolution values into the cfg config instructions, like in the Python script at https://github.com/IntelRealSense/librealsense/issues/6075

kevinbriandIno, 2023-07-27:
Hey @MartyG-RealSense, 
I should've been more precise, that's the json template I'm using. What I've been testing is
"controls-laserpower": "100",
"controls-laserstate": "on"
But when I grab the config after applying it I always get back the default config. So does serialize_json return the default config or the config currently applied? If not would there be a way to confirm the config is applied?
For the resolution we do indeed load it in after. The resolution in this json format was more of a safety net if our standard configuration fails. But thanks for the info didn't know the pipeline start would overwrite it.

MartyG-RealSense, 2023-07-27:
There is not a way to confirm that the json has been successfully applied other than to 'get' the value of a changed option to see if it has been changed from the known default.

If you are only setting the Laser Power value then instead of loading a json, it may be easier to just set its value with Python code using the instruction **rs_option_laser_power**, like the scripting at https://github.com/IntelRealSense/librealsense/issues/9327#issuecomment-872907782

If Laser Power is set to a value greater than 1 then the laser will be On and if its value is set to 0 then the laser will be Off.

kevinbriandIno, 2023-07-28:
So I played around with the configuration and the laser power option seem to work if I set it to on and the power to 0. But setting it to off and having a value on the laserpower seems to ignore the laserState. thanks for the fast return and sorry about the confusion

MartyG-RealSense, 2023-07-28:
If the laser is off and Laser Power is set to a a value greater than 0 than the laser's state will automatically change to On.

MartyG-RealSense, 2023-08-05:
Hi @kevinbriandIno  Do you require further assistance with this case, please?  Thanks!

kevinbriandIno, 2023-08-07:
Sorry about the late response. No you answered my question correctly and the solution was found. thank you!

MartyG-RealSense, 2023-08-07:
No problem at all.  As you have a solution, I will close this case.  Thanks very much for the update!

closed 2023-08-07 by MartyG-RealSense