realsenseai/librealsense #12047: WARNING : (d400-factory.cpp:1195) DS5 group_devices is empty.

jesusramondovale, 2023-07-27:
I get this output when trying to execute 
``` ros2 run realsense2_camera realsense2_camera_node ```
on my DEBIX Model A (2GB RAM) Kernel 5.15 with Ubuntu 22.04 (Debix Image) and ROS2 Humble installed: 

Output: 
```
[INFO] [1690469744.075303649] [camera]: RealSense ROS v4.54.1
[INFO] [1690469744.075683689] [camera]: Built with LibRealSense v2.54.1
[INFO] [1690469744.075766698] [camera]: Running with LibRealSense v2.54.1
 27/07 16:55:44,103 WARNING [281472744417504] (d400-factory.cpp:1195) DS5 group_devices is empty.
 27/07 16:55:44,483 ERROR [281472744417504] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 27/07 16:55:44,504 WARNING [281472744417504] (rs.cpp:312) null pointer passed for argument "device"
[WARN] [1690469744.504959572] [camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[ros2run]: Segmentation fault
```

jesusramondovale, 2023-07-28:
Hi! 

I succesfully updated the camera's firmware version to the last and recommended : 5.15.0.2 but nothing changed ....

```
debix@DEBIX26:~$ sudo rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense D435I
    Serial Number                 :     231122071093
    **Firmware Version              :     5.15.0.2
    Recommended Firmware Version  :     5.15.0.2**
    Physical Port                 :     2-1.1-3
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     221123063041
    Firmware Update Id            :     221123063041
```

But the output when I run 
```ros2 run realsense_camera realsense_camera_node```
is exactly the same :

```
debix@DEBIX26:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1690522784.438752225] [camera]: RealSense ROS v4.54.1
[INFO] [1690522784.439068359] [camera]: Built with LibRealSense v2.54.1
[INFO] [1690522784.439139361] [camera]: Running with LibRealSense v2.54.1
 28/07 07:39:44,466 WARNING [281473406265568] (d400-factory.cpp:1195) DS5 group_devices is empty.
 28/07 07:39:44,851 ERROR [281473406265568] (librealsense-exception.h:52) xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
 28/07 07:39:44,868 WARNING [281473406265568] (rs.cpp:312) null pointer passed for argument "device"
[WARN] [1690522784.868305539] [camera]: Device 1/3 failed with exception: xioctl(VIDIOC_S_EXT_CTRLS) failed Last Error: Invalid argument
[ros2run]: Segmentation fault
```

MartyG-RealSense, 2023-07-28:
It is normal for the Camera Locked value to be Yes. It means that it is not one with special debug functions used by Intel engineers.

The DKMS packages at the distribution_linux.md page will not work with an Arm computer such as your Debix.  They are designed for computers with an x86 / x64 architecture such as desktop and laptop PCs.  An exception to this is Nvidia Jetson Arm-based boards, which have their own dedicated Jetson packages for librealsense.

**DS5 group_devices is empty** indicates that the librealsense and ROS wrapper software are being correctly detected but when the launch process then looks for the camera hardware, it cannot detect it.

jesusramondovale, 2023-07-28:
I don't know how, but ROS camera node is now working, it's publishing on multiple ROS topics such as: 

[image]

But I can't visualize any camera image topic on RQT or Rviz2

jesusramondovale, 2023-07-28:
Don't ask how, but now the realsense2_camera_node is working, publishing multiple topics: 
[image]

But the output of the camera node gives some warnings and errors, and I can't visualize the camera video on ROS 'network': 

```
[realsense2_camera_node-1]  28/07 12:08:00,819 WARNING [281472351004896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 12:08:00,830 WARNING [281472351004896] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 12:08:03,412 ERROR [281471889696992] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130

```

MartyG-RealSense, 2023-07-28:
The `control_transfer returned ` warning **Resource temporarily unavailable** can indicate that there is a communication problem between the camera and the computer.  If the message is only generating a couple of times in the log instead of continuously though then it is safe to ignore it.

jesusramondovale, 2023-07-28:
It's continously. It never stops ... I only copied two of them but there are lots of them messages

MartyG-RealSense, 2023-07-28:
It appears that a lot of streams are enabled at the same time.  Let's try a custom launch where only depth is enabled and it is set to a low FPS speed and resolution to see if it reduces the number of warnings.

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=6400x480x15 enable_color:=false`

jesusramondovale, 2023-07-28:
Same output: 

```
[realsense2_camera_node-1] [INFO] [1690545536.604039116] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1690545536.756401910] [camera.camera]: 
[realsense2_camera_node-1]  28/07 13:58:56,893 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:58:56,903 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:58:56,914 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:58:56,925 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:58:56,935 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:58:56,946 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1] [WARN] [1690545537.486425014] [camera.camera]: XXX Hardware Notification:Motion Module force pause,1.69055e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1690545537.486794148] [camera.camera]: Hardware Notification:Motion Module force pause,1.69055e+12,Error,Hardware Error

[realsense2_camera_node-1]  28/07 13:59:45,083 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,093 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,104 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,114 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,125 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,135 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,146 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[realsense2_camera_node-1]  28/07 13:59:45,156 WARNING [281472409659616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2023-07-28:
It was confirmed in an earlier discussion that the camera works normally on another computer and the problems only occur on your Debix Model A.  And unfortunately you are not able to change to another type of computing board.

Do you have a different USB cable that you can try with the camera to confirm if the Debix is not working well with your current cable even though it works with the camera on your other computer.

jesusramondovale, 2023-07-28:
Yes, I'll try with another USB cable and come back to you. Thanks!

MartyG-RealSense, 2023-08-03:
Apologies for the delay in responding.  A notification is not sent to me if an edit is made to a comment instead of a new comment being posted.

Has there been no change since the previous week?

closed 2023-08-16 by MartyG-RealSense

MartyG-RealSense, 2023-09-27:
Hi @jesusramondovale  I have posted a reply at https://github.com/IntelRealSense/realsense-ros/issues/2816#issuecomment-1736878855