realsenseai/librealsense #12049: Unable to locate package librealsense2-utils

jesusramondovale, 2023-07-28:
I'm following the guide at https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

but when I try to install the librealsense2-dkms package : 

```
debix@DEBIX26:~$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package librealsense2-dkms
```

MartyG-RealSense, 2023-07-28:
Hi @jesusramondovale  As mentioned earlier in your other case at https://github.com/IntelRealSense/librealsense/issues/12047#issuecomment-1655350571 the packages at the **distribution_linux.md** instruction page are designed for computers with an x86 / x64 architecture rather than the Arm architecture of the Debix Model A computing board that you are using.

MartyG-RealSense, 2023-08-03:
Hi @jesusramondovale  Bearing in mind the information in the commenr above, do you have an update about this case that you can provide please?  Thanks!

closed 2023-08-16 by MartyG-RealSense