realsenseai/librealsense #12050: Access infrared projector pattern on D455

vstadnytskyi-FDA, 2023-07-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455} | 
| Firmware Version                   | 2.50.0.3785 | 
| Operating System & Version |   Win 11  |                                | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {AR }                          | 

### Issue Description
Hello,

I would like to access infrared project on D455, specifically, I want to be able to obtain the pattern as an array.

MartyG-RealSense, 2023-07-28:
Hi @vstadnytskyi-FDA  The projector component of the camera does not provide information about the semi-random dot pattern that it projects.  You can turn the projection on and off and increase or decrease its visibility by changing the value of the **Laser Power** setting.  But data about individual points on the pattern cannot be obtained.  

Detailed information about the operatig principles of the projector can be found in Intel's white-paper guide at the link below.

https://dev.intelrealsense.com/docs/projectors

MartyG-RealSense, 2023-08-05:
Hi @vstadnytskyi-FDA  Do you require further assistance with this case, please?  Thanks!

vstadnytskyi-FDA, 2023-08-09:
Thank you for your help!