realsenseai/librealsense #12056: Is this a problem with align?

ioktem, 2023-07-31:
| Required Info                         |       d435                                                         |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> 5.15.0.2) | 
| Operating System & Version |   {Win (10)| 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

In the area I marked as red, depth information cannot be read in the depth photograph. I have the same problem with 2 d435 realsense cameras. This unread area gets bigger if the object gets closer to the camera. How can I fix this.

[image]

[image]
 request / etc..>

ioktem, 2023-07-31:
the error still persists.

Do you think the depth view after calibration is worse than before? I think the area around the whiteboard is starting to look worse.
![n4_depth_Depth](https://github.com/IntelRea
[image]
lSense/librealsense/assets/82447417/81323fbd-846a-44ac-9cac-bc51f79a89de)

MartyG-RealSense, 2023-07-31:
If the black area around the table represents the floor then whilst the floor area looks less clean after calibration, the depth values appear to be more accurate.  If the floor under the table is being observed then there should not be large areas of red, as that would typically represent depth values that are 4 meters or more away from the camera.

The camera is likely having difficulty reading depth information from the floor because it appears to be black or dark grey in color.  It is a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces.  The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.

ioktem, 2023-07-31:
1. The whiteboard is 120 cm from the camera and 140 cm from the floor. The floor is flat carpet.

2.When looking from the back of the camera, the depth information on the left side of the products cannot be read. I calibrated as you said. Can you help me to solve this problem?

ioktem, 2023-08-01:
we thought this(black edge) was a calibration error. 
1. But are you saying that this is normal?  
2. I wonder why the black edge is on the left side?
3. Thank you for helping

MartyG-RealSense, 2023-08-01:
1.  Yes, the silhoutte is normal.  Whilst the ideal is for it to equally surround an object, it is more commonly biased towards the left side of the object.

2.  Depth is generated using the left and right infrared frames, but relies primarily on the left side.  The centerline of the left infrared sensor is also the 0,0,0 origin point of depth.

3.  You are very welcome!

closed 2023-08-16 by MartyG-RealSense