realsenseai/librealsense #12057: RGB exposure settings takes 'forever' until active

Siegel20, 2023-07-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC            |
| SDK Version                            |  2.53.1                          | 
| Language                            | python                          | 
| Segment			|  Robot                          | 

                 

### Issue Description

Setting: 
- RGB @ 30 FPS with 1920 x 1080 (auto-exposure off)
- IR1/2 @ 30  FPS with 1280 x 720 (auto-exposure on, dark environement: typical IR exposure times: 10.000 to 20.000 us)
- depth @ 30 FPS with 1280 x 720

Setting RGB exposure values of 5000 or 10000 using set_option() and retrieving the resulting frame via wait_for_frames() often does not result in a frame where the settings is applied (checked via frame_metadata_value.actual_exposure). 

It takes up to several hundred frames, until the new setting is active (I set it again if the actual value does not match the desired value). 

The setting is applied faster, if I deactivate IR1 and IR2. But I would like to use IR1 and IR2 as well. 

Is this a known issue?

MartyG-RealSense, 2023-07-31:
Hi @Siegel20  The RGB exposure uses a different scale to depth exposure.  RGB exposure uses small values such as 156, whilst depth exposure uses large values such as 33000.  So in most cases RGB exposure should not be set to 5000 or 10000, even though doing so is supported.

In a test that I performed in the RealSense Viewer with RGB exposure of 156 and 10000, at 156 it had an FPS of 30 whilst at 10000 exposure it had an FPS of 0 and a 'No Frames Received' error.

Siegel20, 2023-08-09:
I will try to have enough light so I can avoid getting into these problems. Thank you.

closed 2023-08-09 by Siegel20