realsenseai/librealsense #12063: librealsense2-dkms v1.3.22 does not install on kernel v5.15.0-24-lowlatency

twahrburg-google, 2023-08-01:
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 22.04 | 
| Kernel Version (Linux Only)    | 5.15.0-24-lowlatency  | 
| Platform                                 | PC |

### Issue Description
We're installing librealsense2-dkms in a chroot environment set up for 5.15.0-24-lowlatency.

**librealsense2-dkms v1.3.19** installs, while **librealsense2-dkms v1.3.22** does not. 

Is this a bug, or is support for kernel version 5.15 being deprecated? We've had no issues with the drivers on lowlatency prior.

```
Building for 5.15.0-24-lowlatency
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Building initial module for 5.15.0-24-lowlatency
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/librealsense2-dkms.0.crash'
Error! Application of patch 92-realsense-hid-5.15.patch failed.
Check /var/lib/dkms/librealsense2-dkms/1.3.22/build/ for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 6
Processing triggers for man-db (2.10.2-1) ...
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

ynyBonfennil, 2023-08-02:
It is a bit different environment, but anyway I got the same error `Error! Application of patch 92-realsense-hid-5.15.patch failed.`. My environment is as follow:

- Host machine: Ubuntu 22.04 (kernel version 5.19.0-50-generic)
- Docker container: [ubuntu:jammy](https://hub.docker.com/layers/library/ubuntu/jammy/images/sha256-b060fffe8e1561c9c3e6dea6db487b900100fc26830b9ea2ec966c151ab4c020?context=explore)

After several debuggings, I found that linux header in the docker container was actually `linux-headers-5.15.0-78`, which is different from the kernel version.
So I ran `apt-get install linux-headers-$(uname -r)` and installed `linux-headers-5.19.0-50-generic`. Then `librealsense2-dkms` can be installed without errors.

ynyBonfennil, 2023-08-02:
By the way I had the same error when I was using ubuntu 22.04 server (linux kernel 5.15.0), and in this case I couldn't install librealsense2-dkms even in the host machine. I had to use `librealsense2-dkms v1.3.19` as you described above.

MartyG-RealSense, 2023-08-02:
Thanks so much @ynyBonfennil for kindly offering advice to @twahrburg-google   :)

Hi @twahrburg-google   Does the above advice from @ynyBonfennil solve your problem, please?  Thanks!

twahrburg-google, 2023-08-02:
@MartyG-RealSense Unfortunately, it does not. Thank you for insight though, @ynyBonfennil . I appreciate the testing. It provided useful information.  

We could look at upgrading from Kernel v5.15 to v5.19 if necessary. We'd prefer not deviating too far from our current kernel version to not potentially break other drivers, and libraries. We'll upgrade if v5.15 will no longer be supported for Realsense going forward. We'd appreciate advice on what kernel version to target if we should upgrade our kernel.

MartyG-RealSense, 2023-08-03:
5.15 will continue to be supported.  It is the most recent officially supported kernel at the time of writing this, and support for newer kernels is added in future RealSense SDK releases.  If you are currently using 5.15 then I recommend continuing to do so.

twahrburg-google, 2023-08-09:
Yes. v1.3.22 of librealsense2-dkms v1.3.22 doesn't install on v5.15.0-24-lowlatency. Should I expect this is something that will be resolved in v1.3.23+?

twahrburg-google, 2023-08-11:
I will attempt to build from source.

twahrburg-google, 2023-08-14:
I'm able to build the SDK from source using the  -DFORCE_RSUSB_BACKEND=TRUE flag. Will I be able to build the DKMS module from source?

MartyG-RealSense, 2023-08-15:
DKMS are packages rather than source code, so they cannot be built from source as they are already built.

twahrburg-google, 2023-08-15:
It appears that 92-realsense-hid-5.15.patch is failing to be applied as the files it's attempting to patch has already been patched in v1.3.22

I unpacked the deb files for both librealsense2-dkms_1.3.19-0ubuntu1_all and librealsense2-dkms_1.3.22-0ubuntu1_all

`
 diff  --suppress-common-lines -c  librealsense2-dkms_1.3.19-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c librealsense2-dkms_1.3.22-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c 
*** librealsense2-dkms_1.3.19-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c	2021-10-31 13:53:10.000000000 -0700
--- librealsense2-dkms_1.3.22-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c	2023-05-28 05:04:07.000000000 -0700
***************
*** 280,285 ****
--- 280,286 ----
  			hid_sensor_convert_timestamp(
  					&accel_state->common_attributes,
  					*(int64_t *)raw_data);
+ 		ret = 0;
  	break;
  	default:
  		break;

diff  --suppress-common-lines -c  librealsense2-dkms_1.3.19-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c librealsense2-dkms_1.3.22-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c
*** librealsense2-dkms_1.3.19-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c	2021-10-31 13:53:10.000000000 -0700
--- librealsense2-dkms_1.3.22-0ubuntu1_all/data/usr/src/librealsense2-dkms/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c	2023-05-28 05:04:07.000000000 -0700
***************
*** 231,236 ****
--- 231,237 ----
  		gyro_state->timestamp =
  			hid_sensor_convert_timestamp(&gyro_state->common_attributes,
  						     *(s64 *)raw_data);
+ 		ret = 0;
  	break;
  	default:
  		break;
`

`
cat 92-realsense-hid-5.15.patch 
From b8e6c4a6bf5cfd1064a0d0acdb61e40b7bdf10c9 Mon Sep 17 00:00:00 2001
From: Dmitry Perchanov <dmitry.perchanov@intel.com>
Date: Thu, 27 Oct 2022 18:52:51 +0300
Subject: [PATCH] fix the retval in [accel|gyro]_3d_capture_sample Module
 version realsense2-dkms to track changes Authored by: Yu MENG
 <yu1.meng@intel.com>

---
 5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c | 2 ++
 5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c   | 2 ++
 2 files changed, 4 insertions(+)

diff --git a/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c b/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c
index a2def6f..5df6c69 100644
--- a/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c
+++ b/5.15.0/drivers/iio/accel/hid-sensor-accel-3d.c
@@ -280,6 +280,7 @@ static int accel_3d_capture_sample(struct hid_sensor_hub_device *hsdev,
 			hid_sensor_convert_timestamp(
 					&accel_state->common_attributes,
 					*(int64_t *)raw_data);
+		ret = 0;
 	break;
 	default:
 		break;
@@ -461,3 +462,4 @@ MODULE_DESCRIPTION("HID Sensor Accel 3D");
 MODULE_AUTHOR("Srinivas Pandruvada <srinivas.pandruvada@intel.com>");
 MODULE_LICENSE("GPL");
 MODULE_IMPORT_NS(IIO_HID);
+MODULE_VERSION("realsense2-dkms");
diff --git a/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c b/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c
index 8f0ad02..ba5a536 100644
--- a/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c
+++ b/5.15.0/drivers/iio/gyro/hid-sensor-gyro-3d.c
@@ -231,6 +231,7 @@ static int gyro_3d_capture_sample(struct hid_sensor_hub_device *hsdev,
 		gyro_state->timestamp =
 			hid_sensor_convert_timestamp(&gyro_state->common_attributes,
 						     *(s64 *)raw_data);
+		ret = 0;
 	break;
 	default:
 		break;
@@ -395,3 +396,4 @@ MODULE_DESCRIPTION("HID Sensor Gyroscope 3D");
 MODULE_AUTHOR("Srinivas Pandruvada <srinivas.pandruvada@intel.com>");
 MODULE_LICENSE("GPL");
 MODULE_IMPORT_NS(IIO_HID);
+MODULE_VERSION("realsense2-dkms");
-- 
2.37.1

`

MartyG-RealSense, 2023-08-16:
I will highlight your issue in the coming week to my Intel RealSense colleagues.  In the meantime, the best available option may be to continue with the RSUSB source code version of librealsense or try a different 5.15 kernel version if your project allows it to see whether there is a conflict with your lowlatency kernel version specifically (as 5.15 kernels usually work fine).

twahrburg-google, 2023-08-24:
@MartyG-RealSense Any updates?

twahrburg-google, 2023-08-25:
I was able to get v1.3.22 of the dkms module to install by unpacking the deb package, modifying `./usr/src/librealsense2-dkms-1.3.22/dkms.conf ` line 40:

From:

`PATCH[2]="92-realsense-hid-5.15.patch"`

To:

`PATCH[2]=""`

I then repackaged, and was able to install with dpkg.

```
root@wahrburg:/# mkdir librealsense2-dkms_1.3.22-0ubuntu1_all     
root@wahrburg:/# dpkg-deb -x librealsense2-dkms_1.3.22-0ubuntu1_all.deb librealsense2-dkms_1.3.22-0ubuntu1_all
root@wahrburg:/# dpkg-deb --control librealsense2-dkms_1.3.22-0ubuntu1_all.deb librealsense2-dkms_1.3.22-0ubuntu1_all/DEBIAN
root@wahrburg:/# vim librealsense2-dkms_1.3.22-0ubuntu1_all/usr/src/librealsense2-dkms-1.3.22/dkms.conf 
root@wahrburg:/# dpkg -b librealsense2-dkms_1.3.22-0ubuntu1_all librealsense2-dkms_1.3.22-0ubuntu1_all_mod.deb
dpkg-deb: building package 'librealsense2-dkms' in 'librealsense2-dkms_1.3.22-0ubuntu1_all_mod.deb'.
```

MartyG-RealSense, 2023-08-25:
I had highlighted your issue to my Intel RealSense colleagues.  It's great to hear that you found a solution and I will let my colleagues know.  Thanks very much for the update!

dmipx, 2023-08-28:
Hi @twahrburg-google 
Next time you can tag me so I will be notified.
We submitted HID bugfix to kernel tree and it was applied to all kernels with this bug.
All later minor releases had this bugfix and our dkms failed.
There is no much difference in 1.3.19 and 1.3.22, we updated for newer SKU's and new formats.
We updated this issue and it will be available on version 1.3.24
Regards,
Dmitry Perchanov.

twahrburg-google, 2023-08-28:
@dmipx @MartyG-RealSense Glad to hear. We currently have our workstations pinned to v1.3.19, and will update via APT when 1.3.24 becomes available. I mainly wanted to create this issue to ensure we'd be able to update in the future, and it looks like we will. We didn't noticed much difference with the 1.3.22 hack, so we'll simply wait for the next official release. Thanks so much! :)

MartyG-RealSense, 2023-08-29:
@twahrburg-google  You are very welcome!

MartyG-RealSense, 2023-09-07:
Thanks so much @adamconkey for sharing the solution that worked for you in your situation!

MartyG-RealSense, 2023-09-13:
Hi @twahrburg-google  Do you require further assistance with this case at this time, please?  Thanks!

closed 2023-09-19 by MartyG-RealSense