realsenseai/librealsense #12064: how to trigger D435i camera via MCU

gyt290811, 2023-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

hi,
Do you know how to trigger D435i camera via MCU? We are trying to fix this issue out. According to the official documentation, the camera only facing realsense cameras.
But these don't meet our requirements, so I'd like to ask if you have any solutions for this? For example, modify some places in the SDK so that the camera can be triggered externally。

Thanks!

MartyG-RealSense, 2023-08-02:
Hi @gyt290811  The solution is likely to be to set the camera to use **hardware sync** and define the SDK's **Inter Cam Sync Mode** option to a value of '4'.  This will cause the camera to wait indefinitely until an external trigger pulse is received before initiating a capture.

To achieve this, a wire would be connected from the device that you are using to generate a trigger to **Pin 5 (sync pin)** on the set of hardware sync pins concealed under a small removable cover on the top of the D435i camera casing.

[image]

The trigger pulse must be 1.8 volts in order to be compatible with the D435i's sync pin.  If your trigger is not 1.8 V then you can alter the voltage using an electronics component known as a voltage shifter or level shifter.  RealSense users have used a level shifter similar to the one at the link below.

https://www.sparkfun.com/products/12009

Inter Cam Sync Mode value 4 and above (known as genlock) is no longer supported by Intel but the ability to use it remains in the SDK.  An archived PDF copy of Intel's 'External Syncronization (Genlock)' white-paper guide preserves the contents of that guide for those who need it.   You can download it from the link below.

[External Synchronization of Intel® RealSense™ Depth cameras.pdf](https://github.com/IntelRealSense/librealsense/files/10229019/External.Synchronization.of.Intel.RealSense.Depth.cameras.pdf)

gyt290811, 2023-08-03:
Thank you very much! @MartyG-RealSense 
Based on the information and feedback you provided, we have **successfully completed the external hardware trigger for the depth image** of D435 camera! We carefully reviewed the PDF document you shared, which mentioned that color images do not supporte synchronization . 
However, we would like to ask if **there are any other methods to hardware trigger color images as well**?

MartyG-RealSense, 2023-08-03:
Most RealSense camera models cannot hardware-sync depth and colour because the RGB sensor is not mounted on the main camera circuit board and is instead attached to it via a cable.  The D415 model is the only one where the RGB sensor is integrated on the circuit board.

Even if depth and RGB could be synched with a D415 using Inter Cam Sync Mode '3', it would likely not be suitable for your application.  This is because in mode 3 the camera captures continuously unsynched instead of waiting indefinitely like mode 4 does, until it receives a trigger and performs a synched capture before resuming unsynched capture until the next trigger.

https://github.com/IntelRealSense/librealsense/issues/10926 may be a helpful reference, as it describes how a RealSense user tried to match up genlock-synched depth with unsynched RGB.

closed 2023-08-16 by MartyG-RealSense