realsenseai/librealsense #12069: How to convert depth image to 16-bit (or any desired level of quantization)?

nogaini, 2023-08-03:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.19.0-50-generic | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.1 | 
| Language                            |  Opencv/Python | 
| Segment			|  Others | 

### Issue Description
In [this file](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py) in the repo, I noticed on line 56 that the `cv2.convertScaleAbs()` method is used with `alpha=0.03` for converting the depth image into 8-bit. How can this be modified to quantize the depth image to any level, for instance, 16-bit? 

I believe `alpha` is computed as:= `alpha = ((2**q) - 1)/(max_val - min_val)`, where `q` is the quantization level (e.g. 8, 16, etc.), and `max_val` and `min_val` are the maximum and minimum values respectively in the depth image. For `alpha` to be equal to 0.03 for conversion to 8-bit as in the example file linked above, `max_val` has to be 8500 (since `min_val` is 0). But I'm consistently getting values higher than 8500 in the depth image. So could you tell me why `alpha` is set to 0.03? And what exactly is the formula to compute `alpha` for conversion to other quantization levels?

MartyG-RealSense, 2023-08-04:
Hi @nogaini  I am not familiar with the workings of the cv2.convertScaleAbs function's alpha, so do not know a formula to compute alpha for conversion to other quantization levels.  I will try to provide some hopefully useful information though.

The official OpenCV documentation for cv2.convertScaleAbs describes alpha as being a scale factor.

https://docs.opencv.org/3.4/d2/de8/group__core__array.html#ga3460e9c9f37b563ab9dd550c4d8c4e7d

My research indicates that when using cv2.convertScaleAbs, the alpha and beta parameters are used to adjust the contrast and brightness of an image, with alpha affecting the contrast and beta affecting the brightness.

It appears that 0.03 is considered an appropriate alpha value for converting a 16-bit image to an 8-bit one.

However, a RealSense depth image does not need to be converted to 16-bit as it is already 16-bit by default (its raw pixel depth values are in the **uint16_t** format).  

The pixel depth values of uint16_t can be between 0 and 65535.  The real-world depth in meters can be caculated by multiplying the uint16_t value by the **depth unit scale** of a particular RealSense camera model.  The L515's default depth scale is **0.000250** (whilst the 400 Series D-models mostly use 0.001 as their default scale).  So for example, if the uint16_t value was 8500 then the real world distance in meters on L515 would be 8500 x 0.000250 = **2.125** (meters).

When a 16-bit RealSense image is converted to an OpenCV mat, the range 0-255 is used instead.

chenxiaocongAI, 2023-08-05:
> 

hello, can you help me ?answer my question https://github.com/IntelRealSense/librealsense/issues/12072

MartyG-RealSense, 2023-08-13:
Hi @nogaini  Do you require further assistance with this case, please?  Thanks!

nogaini, 2023-08-14:
Hi @MartyG-RealSense, Sorry about the late reply! Thank you for your assistance, this was really helpful. :) Closing this issue now as it's been resolved.

closed 2023-08-14 by nogaini

MartyG-RealSense, 2023-08-14:
No problem at all, I'm pleased that I was able to help.  Thanks very much for the update!