realsenseai/librealsense #12076: D455 Invalid Value in rs2_get_option_range and Invalid Value in rs2_get_option

immensivevr, 2023-08-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   {Win (10) |                                   
| Platform                                 | PC  |
| SDK Version                            |   2.54   | 
| Segment			|  others                         | 

I can't use camera, and when i open the viewer this give me this 2 errors : 
Invalid Value in rs2_get_option_range(options:000001F754646B70,
option:Depth Units, min:0, max:1, step:1, def:0):
hwmon command 0x2c( 9 1 0 0 ) failed (response -9= No expected user action)

Invalid Value in rs2_get_option(options:000001F754646B70, option:Depth
Units):
hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

and when i try hardware reset  give me this : 
sensor.cpp:661 - acquire_power failed: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The specified path cannot be found."
error-handling.cpp:96 - Error during polling error handler: MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070003: "The specified path cannot be found."

i try whit Intel.Realsense.CustomRW.exe -g
give success, but nothing is change. 
The result :
CustomRW for Intel RealSense D400, Version: 2.13.1.0

  Device PID: 0AD1
  Device name: Intel RealSense D400
  Serial number: ffffffffffff
  Firmware version: 05.15.00.02
  Type: 3.2
  Physical port: \\?\usb#vid_8086&pid_0ad1&mi_00#6&b56e917&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

Calibration on device successfully reset to default gold factory settings.

if i use rs-terminal give me this :

Commands XML file not provided.
you still can send raw data to device in hexadecimal
separated by spaces.
Example GVD command for the SR3XX:
14 00 ab cd 3b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Example GVD command for the D4XX:
14 00 ab cd 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

#>
there are some gvd command for trying to reset ?

if i launch command  Intel.Realsense.CustomRW.exe -r -ex this give me : 
CustomRW for Intel RealSense D400, Version: 2.13.1.0

  Device PID: 0AD1
  Device name: Intel RealSense D400
  Serial number: ffffffffffff
  Firmware version: 05.15.00.02
  Type: 3.2
  Physical port: \\?\usb#vid_8086&pid_0ad1&mi_00#6&b56e917&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 642.883606 641.290710
  PrincipalPointLeft: 648.096863 395.258362
  DistortionLeft: -0.057186 0.065042 -0.001520 -0.000451 -0.021088

  FocalLengthRight: 644.842163 643.287781
  PrincipalPointRight: 633.595459 401.630127
  DistortionRight: -0.058808 0.067831 -0.000607 0.001017 -0.022140

  RotationLeftRight: 0.999903 -0.000022 -0.013903
                     0.000020 1.000000 -0.000153
                     0.013903 0.000152 0.999903
  TranslationLeftRight: -95.042084 0.103101 -0.421945

  HasRGB: 0

Extra calibration information:

World to left rotation matrix (inverse rotation of the left camera in rectified coordinate system):
  0.999955  0.001106  0.009464
  -0.001106  0.999999  0.000075
  -0.009464  -0.000085  0.999955

World to right rotation matrix (inverse rotation of the right camera in rectified coordinate system):
  0.999990  0.001084  -0.004440
  -0.001085  0.999999  -0.000078
  0.004440  0.000082  0.999990

hasRgb is 0, is  not normal right ? there are some commands for writing hasRgb to 1?

MartyG-RealSense, 2023-08-08:
Hi @immensivevr  The serial number ffffffffffff in the CustomRW output indicates that there is a problem.  This is a highly rare one.  At https://github.com/IntelRealSense/librealsense/issues/7337 a RealSense user with this incorrect serial number found that it was caused by damage to the USB cable in their particular case.

Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice?

**hasRgb** should be detected as 1 on a camera model such as D455 which has an RGB sensor.  A value of 0 would indicate that the RGB sensor cannot be detected.  In that case, setting hasRGB to 1 manually would not cause RGB to be detected.

The list of calibration parameters with accompanying correct-looking values suggests that the depth is being detected even if RGB is not.

immensivevr, 2023-08-08:
is there anything i can do with the rs-terminal or other? ?
i not use the official  cable, i use one USB 3.1 gen 1

MartyG-RealSense, 2023-08-08:
The main difference between 3.1 Gen 1 and Gen 2 is that Gen 1 allows speeds of up to 5 Gbps (gigabits per second) whilst Gen 2 allows speeds of up to 10 Gbps.  

Is your cable an old one?  If it is then it could have developed damage over time through bending of the cable or the connector.  If it is a new cable then the quality of the cable can cause problems if it is not high quality or is an inexpensive cable designed for device charging rather than data transfer.

Do you still have the official cable to test the camera with?

immensivevr, 2023-08-09:
in this moment i cant' test with officiale cable, is it possible to buy a replacement with specific features?

MartyG-RealSense, 2023-08-09:
The official cable is not available for separate purchase but a cable of equivalent quality that has been validated for use with RealSense cameras is available from a supplier called **Newnex**.

https://www.newnex.com/realsense-3d-camera-connectivity.php

closed 2023-08-24 by MartyG-RealSense