realsenseai/librealsense #12081: Save and reuse D455 intrinsic and extrinsic parameters

hedeya1980, 2023-08-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 2.48.0.3381 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  C++                          | 
| Segment			|  others                         | 

### Issue Description
I know how to extract intrinsic and extrinsic parameters to use them for projections from RGB to 3D and vice versa. For example, I'm using the following code to extract the parameters from .bag files:

`
    color_intrin = color_profile.as<rs2::video_stream_profile>().get_intrinsics();
    color_extrin_to_depth = color_profile.as<rs2::video_stream_profile>().get_extrinsics_to(depth_profile);
    depth_extrin_to_color = depth_profile.as<rs2::video_stream_profile>().get_extrinsics_to(color_profile);
`

I'm saving individual RGB and PCD files and want to use them instead of applying my pipeline to the .bag files directly.

So, I tried to save the parameters to a json file using the following code:
`
    Json::Value params;

    //depth_intrin = depth_profile.as<rs2::video_stream_profile>().get_intrinsics();
    color_intrin = color_profile.as<rs2::video_stream_profile>().get_intrinsics();
    color_extrin_to_depth = color_profile.as<rs2::video_stream_profile>().get_extrinsics_to(depth_profile);
    depth_extrin_to_color = depth_profile.as<rs2::video_stream_profile>().get_extrinsics_to(color_profile);

    params["color_to_depth_extrin"]["rotation0"] = color_extrin_to_depth.rotation[0];
    params["color_to_depth_extrin"]["rotation1"] = color_extrin_to_depth.rotation[1];
    params["color_to_depth_extrin"]["rotation2"] = color_extrin_to_depth.rotation[2];
    params["color_to_depth_extrin"]["rotation3"] = color_extrin_to_depth.rotation[3];
    params["color_to_depth_extrin"]["rotation4"] = color_extrin_to_depth.rotation[4];
    params["color_to_depth_extrin"]["rotation5"] = color_extrin_to_depth.rotation[5];
    params["color_to_depth_extrin"]["rotation6"] = color_extrin_to_depth.rotation[6];
    params["color_to_depth_extrin"]["rotation7"] = color_extrin_to_depth.rotation[7];
    params["color_to_depth_extrin"]["rotation8"] = color_extrin_to_depth.rotation[8];
    params["color_to_depth_extrin"]["translation0"] = color_extrin_to_depth.translation[0];
    params["color_to_depth_extrin"]["translation1"] = color_extrin_to_depth.translation[1];
    params["color_to_depth_extrin"]["translation2"] = color_extrin_to_depth.translation[2];

    params["color_intrin"]["width"] = color_intrin.width;
    params["color_intrin"]["height"] = color_intrin.height;
    params["color_intrin"]["ppx"] = color_intrin.ppx;
    params["color_intrin"]["ppy"] = color_intrin.ppy;
    params["color_intrin"]["fx"] = color_intrin.fx;
    params["color_intrin"]["fy"] = color_intrin.fy;
    params["color_intrin"]["model"] = color_intrin.model;
    params["color_intrin"]["coeffs0"] = color_intrin.coeffs[0];
    params["color_intrin"]["coeffs1"] = color_intrin.coeffs[1];
    params["color_intrin"]["coeffs2"] = color_intrin.coeffs[2];
    params["color_intrin"]["coeffs3"] = color_intrin.coeffs[3];
    params["color_intrin"]["coeffs4"] = color_intrin.coeffs[4];

    std::cout << "color width: " << color_intrin.width << std::endl;
    std::cout << "color height: " << color_intrin.height << std::endl;
    std::cout << "color ppx: " << color_intrin.ppx << std::endl;
    std::cout << "color ppy: " << color_intrin.ppy << std::endl;
    std::cout << "color fx: " << color_intrin.fx << std::endl;
    std::cout << "color fy: " << color_intrin.fy << std::endl;
    std::cout << "color model: " << color_intrin.model << std::endl;
    std::cout << "color coeffs: " << color_intrin.coeffs[0]<<", " << color_intrin.coeffs[1] << ", " << color_intrin.coeffs[2] << ", " << color_intrin.coeffs[3] << ", " << color_intrin.coeffs[4] << ", "<< std::endl;

    std::cout << "color to depth extr rotation: " << color_extrin_to_depth.rotation[0] << ", " << color_extrin_to_depth.rotation[1] << ", " << color_extrin_to_depth.rotation[2] << ", " << color_extrin_to_depth.rotation[3] << ", " << color_extrin_to_depth.rotation[4] << ", " << color_extrin_to_depth.rotation[5] << ", " << color_extrin_to_depth.rotation[6] << ", " << color_extrin_to_depth.rotation[7] << ", " << color_extrin_to_depth.rotation[8] << std::endl;
    std::cout << "color to depth extr translation: " << color_extrin_to_depth.translation[0] << ", " << color_extrin_to_depth.translation[1] << ", " << color_extrin_to_depth.translation[2] << std::endl;
[... 53 lines cut ...]
    color_extrin_to_depth.rotation[3]= params["color_to_depth_extrin"]["rotation3"].asFloat();
    color_extrin_to_depth.rotation[4]= params["color_to_depth_extrin"]["rotation4"].asFloat();
    color_extrin_to_depth.rotation[5]= params["color_to_depth_extrin"]["rotation5"].asFloat();
    color_extrin_to_depth.rotation[6]= params["color_to_depth_extrin"]["rotation6"].asFloat();
    color_extrin_to_depth.rotation[7]= params["color_to_depth_extrin"]["rotation7"].asFloat();
    color_extrin_to_depth.rotation[8]= params["color_to_depth_extrin"]["rotation8"].asFloat();
    color_extrin_to_depth.translation[0]= params["color_to_depth_extrin"]["translation0"].asFloat();
    color_extrin_to_depth.translation[1]= params["color_to_depth_extrin"]["translation1"].asFloat();
    color_extrin_to_depth.translation[2]= params["color_to_depth_extrin"]["translation2"].asFloat();

    depth_extrin_to_color.rotation[0]= params["depth_to_color_extrin"]["rotation0"].asFloat();
    depth_extrin_to_color.rotation[1]= params["depth_to_color_extrin"]["rotation1"].asFloat();
    depth_extrin_to_color.rotation[2]= params["depth_to_color_extrin"]["rotation2"].asFloat();
    depth_extrin_to_color.rotation[3]= params["depth_to_color_extrin"]["rotation3"].asFloat();
    depth_extrin_to_color.rotation[4]= params["depth_to_color_extrin"]["rotation4"].asFloat();
    depth_extrin_to_color.rotation[5]= params["depth_to_color_extrin"]["rotation5"].asFloat();
    depth_extrin_to_color.rotation[6]= params["depth_to_color_extrin"]["rotation6"].asFloat();
    depth_extrin_to_color.rotation[7]= params["depth_to_color_extrin"]["rotation7"].asFloat();
    depth_extrin_to_color.rotation[8]= params["depth_to_color_extrin"]["rotation8"].asFloat();
    depth_extrin_to_color.translation[0]= params["depth_to_color_extrin"]["translation0"].asFloat();
    depth_extrin_to_color.translation[1]= params["depth_to_color_extrin"]["translation1"].asFloat();
    depth_extrin_to_color.translation[2]= params["depth_to_color_extrin"]["translation2"].asFloat();

`
Here is a link to the json file: https://drive.google.com/file/d/17ke7WhFTpo0LLoSLfwd_wL5RYUPPxCW4/view?usp=sharing

MartyG-RealSense, 2023-08-08:
Hi @hedeya1980  The RealSense SDK does not store calibration variables in json files.  Instead, it saves the values for various non-calibration settings.  Links to examples of such json files can be found at the link below,

https://dev.intelrealsense.com/docs/d400-series-visual-presets#preset-table

[image]

If you know which values you want to set for a particular parameter then you could try setting the values within the program script instead of loading them in from a json file, like in https://github.com/IntelRealSense/librealsense/issues/4061

If you try this and the values still are not the ones that you set in the script then you will know that the problem of the different values is not related to your json file.

hedeya1980, 2023-08-09:
Hello @MartyG-RealSense,

I was asking about the intrinsic and extrinsic parameters (not the calibration parameters), so that I can be able to distribute them with my datasets of RGB and PCD files.
I found the mistake. I didn't write the complete/correct file path in the following line of code:

`
std::ifstream params_file("params.json");
`

Thanks a lot.

closed 2023-08-09 by hedeya1980

MartyG-RealSense, 2023-08-10:
It's excellent to hear that you achieved a solution.  Thanks very much for the update and for sharing your original code and its fix!