realsenseai/librealsense #12087: PointCloud in frame callback for android

Tikitaka02, 2023-08-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Android | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {Java }                          | 
| Segment			|  {Android }                          | 

### Issue Description
Hello I am trying to integrate pointcloud in the frame callback in android as below.

```
private final FrameCallback mFrameHandler = new FrameCallback() {
    @Override
    public void onFrame(final Frame f) {
        FrameSet fs = f.as(Extension.FRAMESET);

        Frame pFrame = pointcloud.process(fs);
        // or
        Frame pFrame = fs.applyFilter(pointcloud).first(StreamType.DEPTH, StreamFormat.XYZ32F);

        Points points = pFrame.as(Extension.POINTS);        
}}

private void configAndStart() throws Exception {
    try (com.intel.realsense.librealsense.Config config = new com.intel.realsense.librealsense.Config()) {
        config.enableStream(StreamType.COLOR, -1, defaultWidth, defaultHeight, StreamFormat.RGBA8, defaultFps);
        config.enableStream(StreamType.DEPTH, defaultWidth, defaultHeight);
        try (PipelineProfile pp = mPipeline.start(config, mFrameHandler)) {
        }
    }
}
```

However, I get out-of-resource and not-supporting error on the points variable. How can I implement pointcloud in frame callback?

MartyG-RealSense, 2023-08-09:
Hi @Tikitaka02  https://github.com/IntelRealSense/librealsense/issues/8680#issuecomment-808876134 demonstrates FrameCallback being imported into a librealsense Android project.

Tikitaka02, 2023-08-09:
Thank you for your reply. I know that the callback is working. Also, if I use the pointcloud with a handler, it works as well. However, the pointcloud with the callback does not work. I am aware that the callback uses some specific process so that the align function also does not work if I remember correctly. I was wondering if this is the case for pointcloud as well.

MartyG-RealSense, 2023-08-09:
Could you try placing the word **callback** into the brackets of your pipe start instruction so that the RealSense SDK knows that it is a callback script, please?

` try (PipelineProfile pp = mPipeline.start(callback, config, mFrameHandler)) {`

MartyG-RealSense, 2023-08-10:
I caried out further research into your question but still could not find any references about using callback with pointcloud on Android, unfortunately.  You may have to use it with a handler instead, as that worked for you.  I do apologize.

MartyG-RealSense, 2023-08-16:
Hi @Tikitaka02  Bearing in mind the information in the comment above, do you require further assistance with this case please?  Thanks!

closed 2023-08-24 by MartyG-RealSense