realsenseai/librealsense #12088: Unclear documentation of Self-Calibration

FHarhammer, 2023-08-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | v2.53.1 | 
| Operating System & Version |   Ubuntu 20.04.6 | 
| Kernel Version (Linux Only)    |  5.15.0-78-generic                                         | 
| Platform                                 | Laptop  |
| SDK Version                            |  2.<?>.<?>                          | 
| Segment			|  Robot                         | 

### Issue Description
I am currently playing around with the Self-Calibration Tools of the Depth Quality Tool and have some question to some confusing information.

1. If I point a D405 camera on the calibration target I get a Subpixel RMS Error of ~0.002 while the documentation says 0.05 is an optimal value so the 0.002 seems to be quite low to me. Could you tell me which value is optimal for a D405?
2. For the plane fit RMS I didn't found any source what is a "good" value, could you tell me what I can expect for this value and when I should do a re-calibration?
3. When I do the On-Chip calibration i sometimes get a positive and sometimes a negative Health Check Value but the "quality scale" in the documentation only shows suggestions for a positive range from 0 to 1. How do i have to interpret these negative values?
4. Also regarding the Health Check Value, does it represent the "Health" before the calibration or after. Somewhere i read it represents the health afterwards but if I apply it and do the calibration again it gets 0 so I guess its the health before the calibration. Am I right with this assumption?
5. When I do the Tare Calibration the documentation says I have to measure the distance from the camera glass to the target and add some offset (3.7mm for D405) to get the distance of the cameras origin. But when I use the "Get" button to get the distance to the target, this distance I get is pretty exactly the distance from the target to the front cover without any offset, so which version is correct now?

I hope you can clear my confusion a bit up.

Thanks in advance

Frederik

MartyG-RealSense, 2023-08-10:
Hi @FHarhammer  Thanks very much for your questions.

1.  The D405 camera model uses a different **depth unit scale** from the rest of the models in the 400 Series stereo depth camera range because it is designed specially for close-range sensing and has a short maximum depth sensing range.  The default depth scale of D405 is **0.01** whilst the default scale of the rest of the 400 Series range is **0.001**.  This likely accounts for the small 0.002 value that you experienced.   

Whilst I do not have a figure for what an ideal subpixel value is for the D405 model, the output of the Depth Quality Tool is very trustable.

 2.  Again, the D405 was released after existing documentation about RMS error was created.  Because RMS error starts at near zero at the camera lenses and increases linearly over meters of distance, the short depth range of the D405 (which has an ideal range of 7 cm to 50 cm) means that RMS error is not likely to be large enough to have a noticable impact on the depth image because the camera will not be far from the observed object / surface.  D405 images are high quality and high accuracy.  

The output of the Depth Quality Tool is highly trustable, but if you wish to check the RMS value yourself then there is a math formula for doing so in point 5 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#verify-performance-regularly-on-a-flat-wall-or-target

3.  There is a detailed discussion about the health check score, including the question of whether the score should be + or -, at https://github.com/IntelRealSense/librealsense/issues/7804

The self-calibration documentation presents the health check value in terms of a positive scale of 0 to 1.  The actual health check score is typically output as a **minus** value though (with -0.03 rated as Good).  The larger the score value is above 0, the less good the calibration is.  So -0.3 would be a worse score than -0.03.

[image]

[image]

4.  The health check score is the score **after** calibration.  That new calibration will not yet have been saved to the camera hardware though.  The aim is to keep repeating the calibration process until you get a score that you are satisfied with, and then save that calibration to the camera.

5.  When performing Tare calibration, using the **Get** button produces the most accurate results.

MartyG-RealSense, 2023-08-16:
Hi @FHarhammer  Do you require further assistance with this case, please?  Thanks!

FHarhammer, 2023-08-16:
No thanks! Thanks for your help!

closed 2023-08-16 by FHarhammer