realsenseai/librealsense #12089: Issue about the color_coordinate

Atphoneee, 2023-08-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

In code :
 "rs2_deproject_pixel_to_point(point_in_color_coordinates, &intr_depth, pixel, dist);"

Can you tell me which physical camera the color_coordinate system is based on?
(It would be even better if it could be labeled on the camera image)

Thanks for your help.

MartyG-RealSense, 2023-08-10:
Hi @Atphoneee  The color coordinate system is based in the **RGB** color sensor of the camera.  On the D435i camera model the RGB sensor is on the end of the right-hand side of the camera when viewing the camera from the front.

[image]

[image]

Atphoneee, 2023-08-11:
Thanks for your help. This information is very useful to me.

MartyG-RealSense, 2023-08-16:
Hi @Atphoneee  Do you require further assistance with this case, please?  Thanks!

Atphoneee, 2023-08-16:
Not really, I'll close the issue later. Thanks again.

closed 2023-08-16 by Atphoneee