realsenseai/librealsense #12093: D405 no image in LabView

HarisSiv, 2023-08-11:
|---------------------------------|------------------------------------------- |
| Camera Model | { D405 } |
| Firmware Version | 05.15.00.02 |
| Operating System & Version | Win 10 |
| Platform | PC |
| SDK Version | V2.53.1 |
| Language | {labview } |
| Segment | others |

### Issue Description
Hello, I have an issue when trying to view images recorded by the D405 depth camera when I am using LabView. 

When I use the Intel realSense Viewer, everything works fine, but when I try to view the same images in LabView, I only see a black screen and the VI is very slow. There are huge delays every time I click.

As can be seen in the linked image, the connection seems to have been established correctly, but I receive no image and I’m not sure what I’m doing wrong.

[image]

I updated the « realsense2.dll » using the DLL of SDK version V2.53.1 and put it in the LabView folder. Since then the VI is able to run, but I still have no image visible.

Do you have any idea what could be the cause of the issues I’m having?

MartyG-RealSense, 2023-08-11:
Hi @HarisSiv  There are known issues with using the LabVIEW wrapper with the D405 camera model, discussed at https://github.com/IntelRealSense/librealsense/issues/11617

The D405 does not have an RGB color sensor and instead obtains RGB from the depth sensor, which can cause complications with programs that were designed for RealSense cameras that have a dedicated RGB sensor.

If you are using SDK 2.53.1 then your firmware version should be **5.14.0.0** instead of 5.15.0.2 (which is designed for use with SDK 2.54.1).  

The RealSense user in the case linked to above found at https://github.com/IntelRealSense/librealsense/issues/11617#issuecomment-1492958159 that they could not get the color stream working because color comes from the depth sensor and so both color and depth had the same stream index number, which is apparently not permitted.

HarisSiv, 2023-08-11:
Hi. I've switched to 5.14.0.0 firmware and am using SDK 2.53.1. I saw the FPS settings end up different for Depth and Color when the VI is put to "run" mode. Is there anything that can be done in order to use the D405 camera in LabView? Or are the cameras supported in the LabView library only Intel cameras with RGB color sensors?

MartyG-RealSense, 2023-08-11:
In the documentation for **Hello World Color + Depth**, it explains how you can change Presets to define which streams are used.  This is useful for using the wrapper with RealSense models that do not have an RGB sensor, such as D410.  

https://dev.intelrealsense.com/docs/labview-wrapper#hello-world-color--depth

If you set the **Face2** preset for depth and the **D410 4** preset for Color then the RGB color image will come from the left IR sensor instead of an RGB sensor.  On D405 the RGB actually does come from the left IR sensor anyway but using the D410 4 color preset may convince the LabVIEW wrapper to display depth + color by treating it like a camera without an RGB sensor.

HarisSiv, 2023-08-14:
Hello, thank you for the help. By trying the different parameters in the documentation you've linked, I succeeded in finding settings which allow an image to be streamed using the D405 in LabView. This issue can be closed.

closed 2023-08-14 by HarisSiv

MartyG-RealSense, 2023-08-14:
That's great to hear.  Thanks very much for the update about your success!