realsenseai/librealsense #12094: Saved RGB frames, depth frames (as png files), and pcd files from the .bag file, and I want to perform projections from 2D to 3D and vice versa

hedeya1980, 2023-08-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 2.48.0.3381 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.0                         | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description

I have no problem to project from 2D to 3D and vice versa when I'm getting the frames directly from the .bag file.

I was following the following steps to get the corresponding 3D point for a 2D point:
```
                    float pixel_tl[2] = { float(x_tl),float(y_tl) };

                    float pixel_distance_in_meters_tl = depth.get_distance(x_tl, y_tl);

                    rs2_deproject_pixel_to_point(Point3d_tl, &color_intrin, pixel_tl, pixel_distance_in_meters_tl);
                    rs2_transform_point_to_point(Point3d_tl_depth, &color_extrin_to_depth, Point3d_tl);
```

I was getting the depth frame as follows:
`
rs2::depth_frame depth = data_aligned_to_color.get_depth_frame();
`

Now, instead of reading directly from the .bag file, I want to make a dataset as individual RGB images (.jpg files), aligned depth images (.png files), and compressed .pcd files, together with the intrinsic and extrinsic parameters; and to do the 2D to 3D projections and vice versa.
[Here is a link](https://drive.google.com/file/d/1876i11qRoWcgpAAaZJF_yh-QpeFZ6V97/view?usp=sharing) for files corresponding to the same frame together with the saved intrinsic and extrinsic parameters (a .json file).

My problem is that I cannot make the get_distance() step, as it has to be applied to a depth frame (rs2::depth_frame).

1. Is it possible to read the depth .png file using cv::imread, and convert it to rs2::depth_frame?
2. I tried to get the distance using the following code, but it gave wrong results:
`
float get_distance_custom(cv::Mat img, int x_d, int y_d, float ppx, float ppy, float fx, float fy)
{
    uint16_t z = img.at<uint16_t>(y_d, x_d); // note that the order is row, column
    float z_meters = z / 1000.0f;
    int x_meters = (x_d - ppx) * z_meters / fx;
    int y_meters = (y_d - ppy) * z_meters / fy;
    return sqrt(pow(x_meters, 2) + pow(y_meters, 2) + pow(z_meters, 2));
}
`
3. Is there any way in which I can go directly from the 2D pixel in the RGB image to the corresponding point in the .pcd file using the intrinsic and extrinsic parameters (without using the depth image)?

Thanks for your help in advance.

MartyG-RealSense, 2023-08-13:
Hi @hedeya1980  Thanks very much for your questions.

1.  A RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2634#issuecomment-433904079 provides an example of C++ code for converting OpenCV's cv::mat format to the RealSense SDK's rs2::frame format. 

2.  https://github.com/IntelRealSense/librealsense/issues/8150 discusses C++ code for obtaining distance from a cv::mat frame.

3.  I am not aware of a method for going directly from the 2D pixel in the RGB image to the corresponding point in a .pcd file, unfortunately.

hedeya1980, 2023-08-14:
Hi @MartyG-RealSense,

Thanks for your answer. 

However, I found that in the first link, the answerers suggested some workarounds, and didn't provide a method to convert from OpenCV's cv::mat format to the RealSense SDK's rs2::frame format.
In the 2nd link, the answerer also didn't provide a method to get the distance from cv::Mat. The provided code snippet was correcting how to get the distance from depth data (not from the image Mat):

`

    uint16_t* dp_data = (uint16_t*)depth.get_data();
    uint16_t dp_pixel = dp_data[100 *  w + 100];

`

So, I still need help on this.

MartyG-RealSense, 2023-08-15:
It looks as though the script at https://github.com/IntelRealSense/librealsense/issues/8150 is indeed calculating the distance value from the cv::mat and not rs2::frame.

First it creates an OpenCV matrix named **image**:

` Mat image(Size(w, h), CV_16U, (void*)depth.get_data(), Mat::AUTO_STEP);`

It works out the **pixel depth** value by using the 'image' mat in the instruction **image.data**:

```
 uint16_t* fr_data = (uint16_t*)image.data;
 uint16_t pixel = fr_data[100,100];
```

Finally, it calculates the mat's real-world distance value in meters by multiplying the pixel depth value **pixel** by the **depth_scale** (depth unit scale) of the RealSense camera that is being used.

If you do not have a RealSense D435i camera attached when running the script then you can change the line `float depth_scale = depth.get_units();` to `float depth_scale = 0.001;` to hard-code the depth unit scale of the D435i into the script instead of retrieving it from the camera in real-time.

hedeya1980, 2023-08-15:
Hi @MartyG-RealSense,
Thanks a lot.
I tried the suggested code, and it worked.
I used the following function:

```

float get_distance_custom(cv::Mat img, int x_d, int y_d)
{
    
    uint16_t* fr_data = (uint16_t*)img.data;
    uint16_t pixel = fr_data[y_d * img.size().width + x_d];
    float z_meters = pixel * 0.001;
    return z_meters;
}
```

But, before that I saved the depth frames from the bag files (through another application) as follows:

```
rs2::depth_frame depth = data_aligned_to_color.get_depth_frame();//.apply_filter(color_map);
Mat depth_image(Size(w, h), CV_16U, (void*)depth.get_data(), Mat::AUTO_STEP);
imwrite(string(argv[6]) + string(argv[7]) + zero_padding(std::to_string(counter), 5) + ".png", depth_image);
   
```

Also, I'll check the additional links you provided above. Thank you so much.

MartyG-RealSense, 2023-08-16:
You are very welcome.  Thanks very much for the update and for sharing your method!

closed 2023-09-27 by MartyG-RealSense