realsenseai/librealsense #12098: WaitForFrames get previous frame

DepapepeOne, 2023-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Win 11  | 
| Platform                                 | PC  |
| Language                            |  C#                           | 
| Segment			|  others                           | 

### Issue Description
I use d455 to take photos against a clock application with milliseconds, and save the rgb pictures. Each picture records the time of taking photos. Why is the clock time in the photos taken earlier than the recorded time? There is a difference of about 100 milliseconds between each photo,  the photo of the previous frame was taken. Is there any way to get the photo taken at roughly the same time as the clock in the photo, here's my code
`var pipe = new Pipeline();

            var cfg = new Config();

            using (var ctx = new Context())
            {
                var devices = ctx.QueryDevices();
                var dev = devices[0];

                Console.WriteLine("\nUsing device 0, an {0}", dev.Info[CameraInfo.Name]);
                Console.WriteLine("    Serial number: {0}", dev.Info[CameraInfo.SerialNumber]);
                Console.WriteLine("    Firmware version: {0}", dev.Info[CameraInfo.FirmwareVersion]);

                var jsonFileName = "D455.json";

                var ser = SerializableDevice.FromDevice(dev); ;

                ser.JsonConfiguration = File.ReadAllText(jsonFileName);

                cfg.EnableStream(Stream.Color, 1280, 720, Format.Bgr8, 30);
            }

            var pp = pipe.Start(cfg);

            using (var swLog = new StreamWriter(@"E:\1\log.txt"))
            {
                while (true)
                {
                    var photoTime = DateTime.Now;

                    using (var frames = pipe.WaitForFrames())
                    {
                        var frameTime = DateTime.Now;

                        var colorFrame = frames.ColorFrame.DisposeWith(frames);

                        var bitmap = new Bitmap(colorFrame.Width, colorFrame.Height, colorFrame.Stride,
                            PixelFormat.Format24bppRgb, colorFrame.Data);

                        var colorImagePath = $"E:\\1\\{photoTime:yyyy_MM_dd_HH_mm_ss_fff}.jpg";
                        bitmap.Save(colorImagePath, ImageFormat.Jpeg);

                        var bmpSaveTime = DateTime.Now;

                        Console.WriteLine("Color saved：" + colorImagePath);

                        var spanTime = bmpSaveTime - photoTime;

                        swLog.WriteLine($"[{photoTime:yyyy-MM-dd HH:mm:ss.fff}]\t[{frameTime:yyyy-MM-dd HH:mm:ss.fff}]\t[{bmpSaveTime:yyyy-MM-dd HH:mm:ss.fff}]\t[{spanTime.TotalMilliseconds}]");
                    }
                }
            }`

The following picture is an example of the above problem. 

Take Photo Time is 2023-08-14 10:23:18.686，but the clock time is 2023-08-14 10:23:18.590
[image]
Take Photo Time is 2023-08-14 10:23:18.714，but the clock time is 2023-08-14 10:23:18.634
[image]

MartyG-RealSense, 2023-08-14:
Hi @DepapepeOne  Each frame from a camera will have a frame number and timestamp that can be queried.  Each camera has its own very precise time clock that is derived from a crystal.

Rather than using DateTime.Now (the computer's system clock), it may be more accurate to retrieve the **timestamp** of each color frame, provided by **FRAME_TIMESTAMP**.  However, the frame timestamp's value will not be in a clock-time format and so may not be suitable for your particular application.

The **WaitForFrames()** instruction blocks until a complete frame is received in order to help to avoid producing incomplete frames.  So I wonder if because you are generating the photoTime value and **then** using WaitForFrames() and taking frameTime, there may be a time offset between photoTime and frameTime because there is a delay due to WaitForFrames waiting for a frame to be completed.

```
var photoTime = DateTime.Now;
using (var frames = pipe.WaitForFrames())
```

You could try moving photoTime under WaitForFrames to avoid this time offset.

```
using (var frames = pipe.WaitForFrames())
var photoTime = DateTime.Now;
```

Another possibility would be to use **PollForFrames()** instead of WaitForFrames(), as PollForFrames() does not block and returns a new frame immediately instead of waiting to ensure that the frame is complete.

DepapepeOne, 2023-08-15:
Thanks for your help

closed 2023-08-24 by MartyG-RealSense