realsenseai/librealsense #12100: Is there Intel.Realsense.dll 32bit?

gjdeorl, 2023-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 2.53.1 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.5                          | 
| Language                            |  C#                          | 

Hello 
I ran into a problem while doing it. When I tried to link with another source, I got an error because of the bits in Intel.Realsense.dll . Maybe the dll file is x86, 32bit? Even if I download the sdk, it's only 64bit.

MartyG-RealSense, 2023-08-14:
Hi @gjdeorl  The Windows version of the RealSense SDK provides both x86 (32 bit) and x64 (64 bit) DLLs.  

If you have installed the full RealSense SDK on Windows using the **Intel.RealSense.SDK-WIN10** installer file then the x86 and x64 DLLs can be found at the folder location below.

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin**

Even though they are within the **Program Files (x86)** directory path, the x64 DLLs should be there.

gjdeorl, 2023-08-15:
[image]
Even after uninstalling the SDK and reinstalling it, x86 is not visible.

MartyG-RealSense, 2023-08-16:
I performed test installations of versions 2.51.1, 2.53.1 and 2.54.1.  I found that the 32-bit folder is installed by Intel.RealSense.SDK-WIN10 up until version **2.51.1** but is not installed in versions after 2.51.1.

gjdeorl, 2023-08-16:
Thank you MartyG~

closed 2023-09-27 by MartyG-RealSense