realsenseai/librealsense #12103: Connection Issues - D457

35selim, 2023-08-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D457 | 
| Firmware Version                   | Cannot connect to RealSense Viewer| 
| Operating System & Version |  Linux (Ubuntu 22.04) | 
| Kernel Version (Linux Only)    |  5.14.0-1056-oem             | 
| Platform                                 | PC |
| SDK Version                            | 2.54.1                          | 
| Language                            |  python                          | 
| Segment			|  Robot                      | 

### Cannot establish the connection
Purchased a D457 device from digikey and it comes in a bag that includes only the Realsense D457 device (no connectors etc.). It is said [here](https://support.intelrealsense.com/hc/en-us/community/posts/10293079498387/comments/14047926505235) that the device can be used with a regular USB-C cable with a performance bottleneck. However, I was not able to establish a connection between D457 and my PC using a USB-C cable even if I flip the cable. How can I solve this problem?

Additionally, since I do not know the contents of the official retail box, can you provide the list of requirements (like connectors) for the device to operate properly with a linux PC. (providing related purchase links would be great)

MartyG-RealSense, 2023-08-14:
Hi @35selim  The D457 has a removable cap on top of its casing.  Under this cap are the hardware sync pins and additionally a 'mode switch' that only the D457 has.  By default this switch is in the **MIPI** position, meaning the camera can only work with a GMSL cable connection.

To enable the camera to be detected in USB mode, the switch should be gently moved aside into the **USB** position using a small object such as a pin.

[image]

35selim, 2023-08-14:
Hi @MartyG-RealSense, thank you for the advice it works! Again, can you provide the list that I mentioned above?

MartyG-RealSense, 2023-08-14:
The D457 retail box should include the camera and a 1 meter GMSL cable for connecting onto the tubular GMSL connector under the large removable cap on the side.  It does not include a USB-C cable.

[image]

Whilst the GMSL cable supplied with the camera is not purchasable separately, it is a generic connector and so any GMSL cable should work with it.

My Intel RealSense colleagues provided technical information about GMSL cabling and a suggested cable brand at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1292320808

In MIPI mode it is necessary to additionally purchase a **de-serializer** component.  Equipment by a company called **Leopard Imaging** will work with D457, as described at https://github.com/IntelRealSense/librealsense/issues/10984#issuecomment-1363694155

A special driver is also required to use the camera in MIPI mode.  Intel provide a reference driver for use with Jetson AGX Xavier computers and Intel's own brand of de-serializer, which is not available for supply to individual customers.  To use the camera in MIPI mode with your own choice of computer, you would need to convert (port) the driver yourself.  Intel are unable to provide assistance with doing so, unfortunately.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Overall it is easiest to use the D457 in USB mode if possible, as you have the same performance and feature-set as a D455 but with the advantage of the IP65 dust and water protection and no need to purchase a de-serializer or port the driver.

35selim, 2023-08-14:
Hi again. Thank you so much!

closed 2023-08-14 by 35selim