realsenseai/librealsense #12112: Not able to install pyrealsense2 on Jetson AGX Orin

sure-explorer, 2023-08-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   |  05.15.00.02  | 
| Operating System & Version |  Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.10.104-tegra      | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                           |  2.39.0                           | 
| Language                               |  python                           | 
| Segment			         |  others                           | 

### Issue Description
I try to develop with D400 camera using python in Jetson AGX Orin.

I have followed https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md to successfully installed the Librealsense2 SDK in the Jetson Ubuntu20.04 (Building from Source using either RSUSB Backend or Native Backend), and the realsense-viewer works just fine.

Then I followed https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 in https://github.com/IntelRealSense/librealsense/issues/11718 to install pyrealsense2, but it end up with "No module named 'pyrealsense2' " in python environment. 

When I check the  /usr/local/lib/python3.8/, there is not any pyrealsense2 there, what should I do, please?

MartyG-RealSense, 2023-08-17:
Hi @sure-explorer  The 2.39.0 version of librealsense is not suitable for the firmware driver version and Ubuntu version that you have.  

Official support for Ubuntu 20.04 Focal was not introduced until librealsense **2.40.0**, the version that came after the 2.39.0 one that you are currently using.

Firmware 5.15.0.2 is designed specifically for use with librealsense **2.54.1**.

I note though that your librealsense installation and the RealSense Viewer appear to be working with your configuration.

********

pyrealsense2 should be able to work with Python 3,8 if pyrealsense2 was installed from source code (which it will have been if the https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 instructions were used).

When pyrealsense2 is installed from source code, its files should be able to be located within the 'build' directory of your librealsense source code folder, at the path **build > wrappers > python**

You can check if your defined PYTHONPATH is correct using the command **which python3**.

sure-explorer, 2023-08-18:
Thank you for responding! 
Now I can import pyrealsense2 in Python, though things are not so expected.

I download the latest Librealsense2 SDK V 2.54.1, I check that my PYTHONPATH is consistent with - DPYTHON_EXECUTABLE param, and the cmake step and make install step are all good without any visible error or warning output.

However, in https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 Step 7."Add these to the end of your .bashrc file":
```
export PATH=$PATH:~/.local/bin
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2
```
If I understand it correctly, under this path `/usr/local/lib/python3.6` there should be a folder named `pyrealsense2` after `make install`, right?
But the fact is that there is no `pyrealsense2` in my path.

As you said, its files should be able to be located within the 'build' directory of my librealsense source code folder, at the path build > wrappers > python, 
Yes indeed, so I just add this path to my PYTHONPATH, till now everything is OK.

MartyG-RealSense, 2023-08-18:
In the .bashrc lines, **python3.6** should be changed to whatever Python 3 version you are using, such as **python3.8**

sure-explorer, 2023-08-18:
Yes, of course, all the python version in this pipeline are all changed to my python version.

Thank you anyway.

Closing it until next error occurs.

MartyG-RealSense, 2023-08-18:
Thank you @sure-explorer