realsenseai/librealsense #12113: Depth Quality Tool Reference

tugbakara, 2023-08-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |       5.4.0-150-generic           | 
| Platform                                 | PC |
| SDK Version                            | 2.50.0          | 
| Language                            |                          | 
| Segment			|  others                           | 

### Issue Description
Hi, I tried may timesto calibrate my two realsensed435i cameras from dynamic calibration tool and I got max these datas below: 
prev value of one camera:

[image]
after calibration:
[image]

but I am not sure that which values better than other? Is there any default value that can be said after this value qualityis bad or good?

MartyG-RealSense, 2023-08-17:
Hi @tugbakara  The easiest measure to use is **Plane-Fit RMS Error**.  RMS Error is where error increases linearly as the distance of an object / surface from the camera lenses increases.  So a smaller RMS value (0.14%) will be better than a larger RMS value (0.18%).

In practice though, the values of those two configurations are so close to each other that choosing either one would be fine.

If you would prefer a clearly stated score for a calibration then the **On-Chip Calibration** tool provides a 'health check score' and a grading such as 'Good' or 'Could be improved' that helps you to decide whether to keep or discard that calibration.  The tool can be accessed from the drop-down menu of the **More** option near the top of the RealSense Viewer's options side-panel.

tugbakara, 2023-08-17:
Thanks Marty I will look health check score:) Have a nice day! :)

MartyG-RealSense, 2023-08-17:
You are very welcome!  The on-chip calibration tool uses a different target image to the Dynamic Calibration tool.  You can find the target and full details of the tool in the guide at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

tugbakara, 2023-08-17:
Actually which calibration method gives more accurate or reliable results?

MartyG-RealSense, 2023-08-17:
They are both accurate and reliable.  On-Chip does not calibrate RGB though, so you should use Dynamic Calibration if you want both the depth and RGB sensors to be calibrated.

On-Chip also has the advantage that you do not need to install a separate calibration software package because it is built into the firmware driver.  You can also activate and control On-Chip calibration with RealSense program scripting.

tugbakara, 2023-08-17:
> If you would prefer a clearly stated score for a calibration then the **On-Chip Calibration** tool provides a 'health check score' and a grading such as 'Good' or 'Could be improved' that helps you [...]

Do I need to make on chip calibration to see helath check?

MartyG-RealSense, 2023-08-17:
Yes, the health check score is provided after the calibration is completed.  The calibration is not automatically saved to the camera though, so you have the choice of discarding the calibration and running it again to try to get a better score.  Then when you are satisified with the score you can save that calibration to the camera.

closed 2023-09-27 by MartyG-RealSense